nanoframework/Home #1262: `nanoff --listtargets` does not display available targets.

networkfusion, 2023-03-31:
### Tool

nanoff

### Description

When the tool nanoff (version 2.4.31) is run with the command `nanoff --listtargets` it does not display the available firmware images available for reference boards, although it does show the community targets.

### How to reproduce

Install the latest version of nanoff
run `nanoff --listtargets`

### Expected behaviour

Shows the reference targets available.

### Screenshots

[image]

stefanomerotta, 2023-03-31:
Maybe I could be wrong but the list seems a merge from the two repositories.
I checked the two repos:
- https://cloudsmith.io/~net-nanoframework/repos/nanoframework-images/groups/
- https://cloudsmith.io/~net-nanoframework/repos/nanoframework-images-community-targets/groups/

The reference targets are all available in the list, while community targets are not all returned, maybe because some are too old and deprecated.
This is the returned list of nanoff --listtargets command where I have highlighted the community targets that can be viewed from the repo.
Some of targets that are referred as community target on site are instead retrieved from the reference target repo.

- ___BrainPad2___
- ESP32_BLE_REV0
- ESP32_BLE_REV3
- ESP32_C3
- ESP32_C3_REV3
- ESP32_GenericDisplay_REV0
- ESP32_LILYGO
- ESP32_OLIMEX
- ESP32_PICO
- ESP32_PSRAM_BLE_GenericGraphic_REV3
- ESP32_PSRAM_REV0
- ESP32_PSRAM_REV3
- ESP32_PSRAM_XTAL26_REV0
- ESP32_REV0
- ESP32_REV3
- ESP_WROVER_KIT
- FEATHER_S2
- ___GHI_FEZ_CERB40_NF___
- ___I2M_ELECTRON_NF___
- ___I2M_OXYGEN_NF___
- KALUGA_1
- M5Core
- M5Core2
- M5StickC
- M5StickCPlus
- MBN_QUAIL
- NETDUINO3_WIFI
- NXP_MIMXRT1060_EVK
- ORGPAL_PALTHREE
- ___PybStick2x___
- SL_STK3701A
- SL_STK3701A_REVB
- ST_B_L475E_IOT01A
- ___ST_NUCLEO144_F412ZG_NF___
- ___ST_NUCLEO144_F439ZI___
- ___ST_NUCLEO144_F746ZG___
- ST_NUCLEO64_F091RC
- ___ST_NUCLEO64_F401RE_NF___
- ___ST_NUCLEO64_F411RE_NF___
- ___ST_STM32F411_DISCOVERY___
- ST_STM32F429I_DISCOVERY
- ___ST_STM32F4_DISCOVERY___
- ST_STM32F769I_DISCOVERY
- ___TI_CC1352P1_LAUNCHXL_868___
- ___TI_CC1352P1_LAUNCHXL_915___
- TI_CC1352R1_LAUNCHXL_868
- TI_CC1352R1_LAUNCHXL_915
- ___WEACT_F411CE___
- WIN32_nanoCLR
- WIN_DLL_nanoCLR

Ellerbach, 2023-04-18:
@networkfusion I've checked and you get all the 50 something targets, both from community and normal repos.
If you are using the `--preview` flag, you indeed get less targets (only the community ones) because we don't have anymore the targets.

I vote for a glitch in the CloudSmith API with a timeout or something like this which prevented you to get the target list.

Shall we close this one?

networkfusion, 2023-04-20:
Yes, let's close. It must have been a glitch.

closed 2023-04-20 by networkfusion