nanoframework/Home #1263: broken build in nanoframework.github.io repo

Arhell, 2023-04-01:
### Description

broken build  https://github.com/nanoframework/nanoframework.github.io

### How to solve the problem

may be https://github.com/dotnet/docfx/releases/tag/v2.64.0
in Breaking Changes

josesimoes, 2023-04-03:
@Arhell FYI: I haven't looked into this with close attention. 
1. Which of the breaking changes do you think may be causing this?
2. Do you have a proposal for a fix?

Arhell, 2023-04-03:
1. Unfortunately it is necessary to test.
2. Alas, no, I installed this version locally and assembled it, everything works.

Ellerbach, 2023-04-04:
> 2\. Alas, no, I installed this version locally and assembled it, everything works.

That seems weird, if works locally, will work in the pipeline, I don't see a difference there 🤪

closed 2023-04-04 by Arhell