nanoframework/Home #1264: Migrate MVVM light to MVVM Toolkit

josesimoes, 2023-04-04:
### Details about Problem

**nanoFramework area:** (Visual Studio extension)

### Description

Currently our VS extension is using MVVM Light, wich is deprecated. The recommended replacement is the [MVVM Toolkit](https://learn.microsoft.com/en-us/dotnet/communitytoolkit/mvvm/).

Migration guide [here](https://learn.microsoft.com/en-us/dotnet/communitytoolkit/mvvm/migratingfrommvvmlight)
And several other docs and samples available.

torbacz, 2023-04-04:
@josesimoes I can do it, I've got plenty of experience with WPF and MVVM

josesimoes, 2023-04-04:
@torbacz awesome!! Thanks for offering. 👍🏻

josesimoes, 2023-04-04:
@torbacz this has become critical!
I was trying to fix a issue with the extension not loading properly and it seems that the issue is comming from GalaSoft.MvvmLight

activity log from VS

```
SetSite failed for package [Visual Studio 2022 extension for .NET nanoFramework. Enables creating C# Solutions to be deployed to a target board and provides debugging tools.]Source: &apos;nanoFramework.Tools.VS2022.Extension&apos; Description: Could not load file or assembly &apos;GalaSoft.MvvmLight, Version=5.4.1.0, Culture=neutral, PublicKeyToken=0ffbc31322e9d308&apos; or one of its dependencies. O sistema n&#x00E3;o conseguiu localizar o ficheiro especificado.
System.IO.FileNotFoundException: Could not load file or assembly &apos;GalaSoft.MvvmLight, Version=5.4.1.0, Culture=neutral, PublicKeyToken=0ffbc31322e9d308&apos; or one of its dependencies. O sistema n&#x00E3;o conseguiu localizar o ficheiro especificado.
File name: &apos;GalaSoft.MvvmLight, Version=5.4.1.0, Culture=neutral, PublicKeyToken=0ffbc31322e9d308&apos;
   at nanoFramework.Tools.VisualStudio.Extension.NanoFrameworkPackage.&lt;InitializeAsync&gt;d__81.MoveNext()
   at System.Runtime.CompilerServices.AsyncTaskMethodBuilder.Start[TStateMachine](TStateMachine&amp; stateMachine)
   at nanoFramework.Tools.VisualStudio.Extension.NanoFrameworkPackage.InitializeAsync(CancellationToken cancellationToken, IProgress`1 progress)
   at Microsoft.VisualStudio.Shell.AsyncPackage.&lt;&gt;c__DisplayClass21_0.&lt;&lt;Microsoft-VisualStudio-Shell-Interop-IAsyncLoadablePackageInitialize-Initialize&gt;b__1&gt;d.MoveNext()
--- End of stack trace from previous location where exception was thrown ---
   at System.Runtime.ExceptionServices.ExceptionDispatchInfo.Throw()
   at Microsoft.VisualStudio.Services.VsTask.RethrowException(AggregateException e)
   at Microsoft.VisualStudio.Services.VsTask.InternalGetResult(Boolean ignoreUIThreadCheck)

=== Pre-bind state information ===
LOG: DisplayName = GalaSoft.MvvmLight, Version=5.4.1.0, Culture=neutral, PublicKeyToken=0ffbc31322e9d308&#x000A; (Fully-specified)
LOG: Appbase = file:///C:/Program Files/Microsoft Visual Studio/2022/Enterprise/Common7/IDE/
LOG: Initial PrivatePath = NULL
Calling assembly : nanoFramework.Tools.VS2022.Extension, Version=2022.3.0.28, Culture=neutral, PublicKeyToken=null.
===
LOG: This bind starts in default load context.
LOG: Using application configuration file: C:\Users\Jos&#x00E9;Sim&#x00F5;es\AppData\Local\Microsoft\VisualStudio\17.0_85b18ea3Exp\devenv.exe.config
LOG: Using host configuration file: 
LOG: Using machine configuration file from C:\Windows\Microsoft.NET\Framework64\v4.0.30319\config\machine.config.
LOG: Post-policy reference: GalaSoft.MvvmLight, Version=5.4.1.0, Culture=neutral, PublicKeyToken=0ffbc31322e9d308
LOG: Attempting download of new URL file:///C:/Program Files/Microsoft Visual Studio/2022/Enterprise/Common7/IDE/GalaSoft.MvvmLight.DLL.
LOG: Attempting download of new URL file:///C:/Program Files/Microsoft Visual Studio/2022/Enterprise/Common7/IDE/GalaSoft.MvvmLight/GalaSoft.MvvmLight.DLL.
LOG: Attempting download of new URL file:///C:/Program Files/Microsoft Visual Studio/2022/Enterprise/Common7/IDE/PublicAssemblies/GalaSoft.MvvmLight.DLL.
LOG: Attempting download of new URL file:///C:/Program Files/Microsoft Visual Studio/2022/Enterprise/Common7/IDE/PublicAssemblies/GalaSoft.MvvmLight/GalaSoft.MvvmLight.DLL.
LOG: Attempting download of new URL file:///C:/Program Files/Microsoft Visual Studio/2022/Enterprise/Common7/IDE/PrivateAssemblies/GalaSoft.MvvmLight.DLL.
LOG: Attempting download of new URL file:///C:/Program Files/Microsoft Visual Studio/2022/Enterprise/Common7/IDE/PrivateAssemblies/GalaSoft.MvvmLight/GalaSoft.MvvmLight.DLL.
LOG: Attempting download of new URL file:///C:/Program Files/Microsoft Visual Studio/2022/Enterprise/Common7/IDE/CommonExtensions/Microsoft/TestWindow/GalaSoft.MvvmLight.DLL.
LOG: Attempting download of new URL file:///C:/Program Files/Microsoft Visual Studio/2022/Enterprise/Common7/IDE/CommonExtensions/Microsoft/TestWindow/GalaSoft.MvvmLight/GalaSoft.MvvmLight.DLL.
LOG: Attempting download of new URL file:///C:/Program Files/Microsoft Visual Studio/2022/Enterprise/Common7/IDE/CommonExtensions/Platform/Debugger/GalaSoft.MvvmLight.DLL.
LOG: Attempting download of new URL file:///C:/Program Files/Microsoft Visual Studio/2022/Enterprise/Common7/IDE/CommonExtensions/Platform/Debugger/GalaSoft.MvvmLight/GalaSoft.MvvmLight.DLL.
LOG: Attempting download of new URL file:///C:/Program Files/Microsoft Visual Studio/2022/Enterprise/Common7/IDE/PrivateAssemblies/DataCollectors/GalaSoft.MvvmLight.DLL.
LOG: Attempting download of new URL file:///C:/Program Files/Microsoft Visual Studio/2022/Enterprise/Common7/IDE/PrivateAssemblies/DataCollectors/GalaSoft.MvvmLight/GalaSoft.MvvmLight.DLL.
LOG: Attempting download of new URL file:///C:/Program Files/Microsoft Visual Studio/2022/Enterprise/Common7/IDE/PrivateAssemblies/DataCollectors/x64/GalaSoft.MvvmLight.DLL.
LOG: Attempting download of new URL file:///C:/Program Files/Microsoft Visual Studio/2022/Enterprise/Common7/IDE/PrivateAssemblies/DataCollectors/x64/GalaSoft.MvvmLight/GalaSoft.MvvmLight.DLL.
LOG: Attempting download of new URL file:///C:/Program Files/Microsoft Visual Studio/2022/Enterprise/Common7/IDE/GalaSoft.MvvmLight.EXE.
LOG: Attempting download of new URL file:///C:/Program Files/Microsoft Visual Studio/2022/Enterprise/Common7/IDE/GalaSoft.MvvmLight/GalaSoft.MvvmLight.EXE.
[... cut ...]
LOG: Attempting download of new URL file:///C:/Program Files/Microsoft Visual Studio/2022/Enterprise/Common7/IDE/PrivateAssemblies/DataCollectors/GalaSoft.MvvmLight/GalaSoft.MvvmLight.EXE.
LOG: Attempting download of new URL file:///C:/Program Files/Microsoft Visual Studio/2022/Enterprise/Common7/IDE/PrivateAssemblies/DataCollectors/x64/GalaSoft.MvvmLight.EXE.
LOG: Attempting download of new URL file:///C:/Program Files/Microsoft Visual Studio/2022/Enterprise/Common7/IDE/PrivateAssemblies/DataCollectors/x64/GalaSoft.MvvmLight/GalaSoft.MvvmLight.EXE.
```

torbacz, 2023-04-04:
Ok, Will look at it tomorrow evening.

iss0134, 2023-04-05:
我是初学者，也遇到了这个问题，如果是第一次创建nanoframework项目，vs2022会出现这个错误。vs2019不会出现这个问题

AllonisDave, 2023-04-06:
Same issues here. Is there a workaround to get this to work in VS2022 or just wait?

torbacz, 2023-04-06:
Still working on fix, no idea if there is a workaround.

iss0134, 2023-04-07:
> Same issues here. Is there a workaround to get this to work in VS2022 or just wait?

VS2019~~~

josesimoes, 2023-04-07:
Right now, for VS2022, the last known good version is 2022.3.0.22

torbacz, 2023-04-07:
@josesimoes Migration completed, but it doesn't solve the issue. I've got the same error with MVVM toolkit.... I've tried to debug and enabled FusionLog for .net, but haven't found any clue what is wrong.

<description>SetSite failed for package [Visual Studio 2022 extension for .NET nanoFramework. Enables creating C# Solutions to be deployed to a target board and provides debugging tools.]Source: &apos;nanoFramework.Tools.VS2022.Extension&apos; **Description: Could not load file or assembly &apos;CommunityToolkit.Mvvm, Version=8.1.0.0, Culture=neutral, PublicKeyToken=4aff67a105548ee2&apos; or one of its dependencies.** The system cannot find the file specified.&#x000D;&#x000A;System.IO.FileNotFoundException: Could not load file or assembly &apos;CommunityToolkit.Mvvm, Version=8.1.0.0, Culture=neutral, PublicKeyToken=4aff67a105548ee2&apos; or one of its dependencies. The system cannot find the file specified.&#x000D;&#x000A;File name: &apos;CommunityToolkit.Mvvm, Version=8.1.0.0, Culture=neutral, PublicKeyToken=4aff67a105548ee2&apos;&#x000D;&#x000A;   at nanoFramework.Tools.VisualStudio.Extension.NanoFrameworkPackage.&lt;InitializeAsync&gt;d__81.MoveNext()&#x000D;&#x000A;   at System.Runtime.CompilerServices.AsyncTaskMethodBuilder.Start[TStateMachine](TStateMachine&amp; stateMachine)&#x000D;&#x000A;   at nanoFramework.Tools.VisualStudio.Extension.NanoFrameworkPackage.InitializeAsync(CancellationToken cancellationToken, IProgress`1 progress)&#x000D;&#x000A;   at Microsoft.VisualStudio.Shell.AsyncPackage.&lt;&gt;c__DisplayClass21_0.&lt;&lt;Microsoft-VisualStudio-Shell-Interop-IAsyncLoadablePackageInitialize-Initialize&gt;b__1&gt;d.MoveNext()&#x000D;&#x000A;--- End of stack trace from previous location where exception was thrown ---&#x000D;&#x000A;   at System.Runtime.ExceptionServices.ExceptionDispatchInfo.Throw()&#x000D;&#x000A;   at Microsoft.VisualStudio.Services.VsTask.RethrowException(AggregateException e)&#x000D;&#x000A;   at Microsoft.VisualStudio.Services.VsTask.InternalGetResult(Boolean ignoreUIThreadCheck)&#x000D;&#x000A;&#x000D;&#x000A;WRN: Assembly binding logging is turned OFF.&#x000D;&#x000A;To enable assembly bind failure logging, set the registry value [HKLM\Software\Microsoft\Fusion!EnableLog] (DWORD) to 1.&#x000D;&#x000A;Note: There is some performance penalty associated with assembly bind failure logging.&#x000D;&#x000A;To turn this feature off, remove the registry value [HKLM\Software\Microsoft\Fusion!EnableLog].&#x000D;&#x000A;</description>

AllonisDave, 2023-04-07:
Is this applicable here?
CommunityToolKit.mvvm doesn’t support any version of .NET Framework: https://www.nuget.org/packages/CommunityToolkit.Mvvm/#supportedframeworks-body-tab

josesimoes, 2023-04-10:
> Is this applicable here? CommunityToolKit.mvvm doesn’t support any version of .NET Framework: https://www.nuget.org/packages/CommunityToolkit.Mvvm/#supportedframeworks-body-tab

We should be good with using it on the VS extension. I've tested it and it looks OK. Let's wait for the release from the pipeline for the definitive proof. 

(@AllonisDave BTW, thank you for your concern and looking into this)

closed 2023-04-10 by josesimoes