nanoframework/Home #1267: Device Explorer Report Error

SoaSoftNet, 2023-04-10:
### Tool

Visual Studio extension

### Description

[image]
VS2022  Device Explorer

AllonisDave, 2023-04-10:
Same here. Basically  "Could not load file or assembly "CommunityToolkit.Mvvm, Version=8.1.0.0" Here is my Activity Log.

[ActivityLog.zip](https://github.com/nanoframework/Home/files/11192701/ActivityLog.zip)

networkfusion, 2023-04-10:
https://github.com/nanoframework/Home/issues/1264#issuecomment-1501681364

AllonisDave, 2023-04-10:
This is using the newest v2022.3.0.29 release

networkfusion, 2023-04-11:
As a workaround, please download and use the last known working extension: https://github.com/nanoframework/nf-Visual-Studio-extension/releases/tag/v2022.3.0.22

SoaSoftNet, 2023-04-13:
[image]
[image]
[image]

pt2go, 2023-04-13:
v2022.3.0.26 can work,but no logo

josesimoes, 2023-04-13:
Fixed with v2022.3.040

closed 2023-04-13 by josesimoes