nanoframework/Home #1270: ST_STM32F769I_DISCO ChibiOS build warnings.

networkfusion, 2023-04-11:
### Target name(s)

ST_STM32F769I_DISCO

### Firmware version

latest

### Was working before? On which version?

?

### Device capabilities

N/A

### Description

When building the firmware in `MinSizeRel` there is a warning related to use of the heap.

```
[build] [732/744] Building C object targets/ChibiOS/ST_STM32F769I_DISCOVERY/CMakeFiles/NF_Network.dir/sources/ChibiOs/ext/lwip/src/api/netdb.c.obj
[build] [733/744] Building C object targets/ChibiOS/ST_STM32F769I_DISCOVERY/CMakeFiles/NF_Network.dir/sources/ChibiOs/ext/lwip/src/api/tcpip.c.obj
[build] [734/744] Building C object targets/ChibiOS/ST_STM32F769I_DISCOVERY/CMakeFiles/NF_Network.dir/sources/ChibiOs/ext/lwip/src/api/netifapi.c.obj
[build] [735/744] Building C object targets/ChibiOS/ST_STM32F769I_DISCOVERY/CMakeFiles/NF_Network.dir/sources/ChibiOs/os/various/evtimer.c.obj
[build] [736/744] Building C object targets/ChibiOS/ST_STM32F769I_DISCOVERY/CMakeFiles/NF_Network.dir/__/_Lwip/nf_sys_arch.c.obj
[build] [737/744] Building C object targets/ChibiOS/ST_STM32F769I_DISCOVERY/CMakeFiles/NF_Network.dir/__/_Lwip/nf_api_msg.c.obj
[build] [738/744] Building C object targets/ChibiOS/ST_STM32F769I_DISCOVERY/CMakeFiles/NF_Network.dir/__/_Lwip/nf_lwipthread.c.obj
[build] [739/744] Building C object targets/ChibiOS/ST_STM32F769I_DISCOVERY/CMakeFiles/NF_Network.dir/__/_Lwip/nf_sockets.c.obj
[build] [740/744] Building C object targets/ChibiOS/ST_STM32F769I_DISCOVERY/CMakeFiles/NF_Network.dir/sources/ChibiOs/ext/lwip/src/netif/ethernet.c.obj
[build] [741/744] Building C object targets/ChibiOS/ST_STM32F769I_DISCOVERY/CMakeFiles/NF_Network.dir/sources/ChibiOs/ext/lwip/src/netif/slipif.c.obj
[build] [742/744] Building C object targets/ChibiOS/ST_STM32F769I_DISCOVERY/CMakeFiles/NF_Network.dir/sources/ChibiOs/ext/lwip/src/apps/sntp/sntp.c.obj
[build] [743/744] Linking CXX static library targets/ChibiOS/ST_STM32F769I_DISCOVERY/libNF_Network.a
[build] [744/744] Linking CXX executable targets/ChibiOS/ST_STM32F769I_DISCOVERY/nanoCLR.elf
[build] /sources/ChibiOs/os/oslib/include/chmemheaps.h:82:7: warning: type 'union heap_header' violates the C++ One Definition Rule [-Wodr]
[build]    82 | union heap_header {
[build]       |       ^
[build] /sources/ChibiOs/os/oslib/include/chmemheaps.h:82:7: note: a different type is defined in another translation unit
[build]    82 | union heap_header {
[build]       |       ^
[build] /sources/ChibiOs/os/oslib/include/chmemheaps.h:86:5: note: the first difference of corresponding definitions is field 'YOU_SHALL_NOT_USE_free'
[build]    86 |   } free;
[build]       |     ^
[build] /sources/ChibiOs/os/oslib/include/chmemheaps.h:86:5: note: a field with different name is defined in another translation unit
[build]    86 |   } free;
[build]       |     ^
[build] /sources/ChibiOs/os/oslib/include/chmemheaps.h:96:8: warning: type 'struct memory_heap' violates the C++ One Definition Rule [-Wodr]
[build]    96 | struct memory_heap {
[build]       |        ^
[build] /sources/ChibiOs/os/oslib/include/chmemheaps.h:96:8: note: a different type is defined in another translation unit
[build]    96 | struct memory_heap {
[build]       |        ^
[build] /sources/ChibiOs/os/oslib/include/chmemheaps.h:99:25: note: the first difference of corresponding definitions is field 'header'
[build]    99 |   heap_header_t         header;     /**< @brief Free blocks list header.    */
[build]       |                         ^
[build] /sources/ChibiOs/os/oslib/include/chmemheaps.h:99:25: note: a field of same name but different type is defined in another translation unit
[build]    99 |   heap_header_t         header;     /**< @brief Free blocks list header.    */
[build]       |                         ^
[build] /sources/ChibiOs/os/oslib/include/chmemheaps.h:96:8: note: type 'union heap_header_t' itself violates the C++ One Definition Rule
[build]    96 | struct memory_heap {
[build]       |        ^
[build] lto-wrapper: warning: using serial compilation of 10 LTRANS jobs
[build] lto-wrapper: note: see the '-flto' option documentation for more information
```

The link to the currently used `chmemheaps` is https://github.com/ChibiOS/ChibiOS/blob/stable_21.11.x/os/oslib/include/chmemheaps.h

### Expected behaviour

No warnings.

### Screenshots

[image]

### Aditional information

Further investigation ongoing.

josesimoes, 2023-04-11:
This is ChibiOS code, better report to Giovanni.

networkfusion, 2023-04-11:
I am just trying to make sure it is not a porting issue against the latest branch before that... but will do once I have investigated enough to know what I am talking about...

networkfusion, 2023-04-25:
After a lot of trial and error, I have tracked this down to being an issue with the OneWire library. Given it is a custom overlay driver for ChibiOS, it is a problem with our code.

Turning off OneWire in CMakePresets.json provides a workaround as long as you dont need OneWire capabilities.

networkfusion, 2025-04-06:
Closing this as I have not seen it recently.

closed 2025-04-06 by networkfusion