nanoframework/Home #1277: Unable to connect to wifi CLR_E_INVALID_OPERATION

jasonkaisersmith, 2023-04-21:
### Library/API/IoT binding

nanoFramework.Networking

### Visual Studio version

VS2022

### .NET nanoFramework extension version

2022.3.0.41

### Target name(s)

ESP_WROVER_KIT

### Firmware version

1.8.1.148

### Description

Seems to be potentially the same issue as reported here:
https://github.com/nanoframework/Home/issues/1166

But that was closed.

Output

    ++++ Exception System.InvalidOperationException - CLR_E_INVALID_OPERATION (1) ++++
    ++++ Message: 
    ++++ System.Device.Wifi.WifiAdapter::NativeScanAsync [IP: 0000] ++++
    ++++ System.Device.Wifi.WifiAdapter::ScanAsync [IP: 000d] ++++
    ++++ nanoFramework.Networking.WifiNetworkHelper::ScanAndConnectDhcp [IP: 000d] ++++
    ++++ WebRequest.Program::Main [IP: 0021] ++++

### Sample project or code

            bool success;
            CancellationTokenSource cs = new(60000);

            success = WifiNetworkHelper
                .ConnectDhcp(MySsid, MyPassword, requiresDateTime: true, token: cs.Token);

alberk8, 2023-04-22:
May I know which firmware was installed on the ESP32?

jasonkaisersmith, 2023-04-22:
Firmware version
ESP32 (ESP32-D0WD-V3 (revision 3))
1.8.1.148

alberk8, 2023-04-22:
> Firmware version ESP32 (ESP32-D0WD-V3 (revision 3)) 1.8.1.148

try install firmware with `nanoff --target ESP32_REV3 --update --serialport COM6`