nanoframework/Home #1284: GCC12 lto-wrapper: warning

networkfusion, 2023-04-26:
### Target name(s)

ST_STM32F769I_DISCO

### Firmware version

GCC12.x

### Was working before? On which version?

GCC 11.x

### Device capabilities

N/A

### Description

With GCC12, a new warning is present.

This warning will only be shown when building "Optimised" firmware (MinSizeRel, Release).
 
This is known and related to the latest `binutils`.

```
[build] [663/664] Linking CXX static library targets/ChibiOS/ST_STM32F769I_DISCOVERY/libNF_Network.a
[build] [664/664] Linking CXX executable targets/ChibiOS/ST_STM32F769I_DISCOVERY/nanoCLR.elf
[build] lto-wrapper: warning: using serial compilation of 9 LTRANS jobs
[build] lto-wrapper: note: see the '-flto' option documentation for more information
[build] /usr/local/bin/gcc/bin/../lib/gcc/arm-none-eabi/12.2.1/../../../../arm-none-eabi/bin/ld: warning: targets/ChibiOS/ST_STM32F769I_DISCOVERY/nanoCLR.elf has a LOAD segment with RWX permissions
```

### How to reproduce

Build firmware with MinSizeRelease

### Expected behaviour

No warnings

### Aditional information

I have tried changing all instances of `-flto` to `-flto=auto` but that does not fix it.

The following thead may help: https://stackoverflow.com/questions/72218980/gcc-v12-1-warning-about-serial-compilation

softopti, 2023-08-10:
I have the same problem with GCC 12.3 and binutils 2.41

networkfusion, 2023-09-29:
Seems to have been fixed with move to GCC 13. Will re-open if required.

closed 2023-09-29 by networkfusion

francoisdepicciotto-tagheuer, 2024-01-22:
I have the same problem on Windows with gcc-13.2.0-mingw-w64ucrt-11.0.1-r4, compiling fortran code (I have no idea if there's a potential link).

I have tried changing all instances of -flto to -flto=auto but that does not fix it.

oppiet30, 2024-04-28:
It does still show up in:
[oppie@dell src]$ gcc --version
gcc (GCC) 14.0.1 20240411 (Red Hat 14.0.1-0)
Copyright (C) 2024 Free Software Foundation, Inc.
This is free software; see the source for copying conditions.  There is NO
warranty; not even for MERCHANTABILITY or FITNESS FOR A PARTICULAR PURPOSE.
 
The warning is still there.  I didn't use the -flto=auto switch. I get the error with just the -flto.

oppiet30, 2024-04-28:
With the -flto=auto makes the warning not show up.

Challanger524, 2025-08-19:
To suppress `lto-wrapper: warning: using serial compilation of N LTRANS jobs` - `-O0` (default) (or no `-O*` at all) optimizations should be passed, `=auto` is not necessary (in my case)