nanoframework/Home #1290: No suitable test runtime provider was found

jasonkaisersmith, 2023-05-01:
### Library/API/IoT binding

nanoFramework.TestFramework

### Visual Studio version

VS2022  17.5.4

### .NET nanoFramework extension version

2022.3.0.41

### Target name(s)

ESP32

### Firmware version

N/A

### Device capabilities

N/A

### Description

When creating a new Unit Test project, it is not possible to run any tests.

There is 1 warning, the output of which states:

========== Starting test discovery ==========
No suitable test runtime provider was found for any source in this run.
Skipping source: C:\Users\Jason\source\repos\nano\GpioCollection\GpioCollection\bin\Debug\GpioCollection.dll (.NETnanoFramework,Version=v1.0, X64)

I have followed the instructions here:
https://docs.nanoframework.net/content/unit-test/running-tests.html

The test(s) are not visible in the test explorer.  Trying to run them results in nothing happening.  After which, the "Run" and "Debug" test options are all greyed out (in Test explorer & solution explorer)

And set:
<TargetFrameworkVersion>Framework40</TargetFrameworkVersion>

My nano.runsettings are:
<?xml version="1.0" encoding="utf-8"?>
<RunSettings>
   
   <RunConfiguration>
	   <MaxCpuCount>1</MaxCpuCount>
       <ResultsDirectory>.\TestResults</ResultsDirectory>
       <TestSessionTimeout>120000</TestSessionTimeout>
	   <TargetFrameworkVersion>Framework40</TargetFrameworkVersion>
       <TargetPlatform>x64</TargetPlatform>
   </RunConfiguration>
   <nanoFrameworkAdapter>
       <Logging>None</Logging> 
       <IsRealHardware>False</IsRealHardware>
       <RealHardwarePort>COM3</RealHardwarePort>
       <CLRVersion></CLRVersion>
       <PathToLocalCLRInstance></PathToLocalCLRInstance>
   </nanoFrameworkAdapter>
</RunSettings>

BTW:  The "Samples" NF Unit test, has a different value set for the Framework version (net48).  But even setting this, cleaning, rebuilding, closing and restarting VS makes no difference and I get exactly the same error.

### How to reproduce

1.  Create a new Unit Test with the NFUnit Test framework.
2.  Set framework version according to documentation.
3. Clean, Rebuild
4. Save and close VS
5. Open VS

networkfusion, 2023-05-01:
Please try manually selecting the `runsettings` and see what happends...

[image]

jasonkaisersmith, 2023-05-02:
After selecting the runsettings manually, I had to restart VS and then the tests were recognised and would run.
Thanks.

closed 2023-05-02 by jasonkaisersmith