nanoframework/Home #1295: UTC offset not output in datetime string

jasonkaisersmith, 2023-05-07:
### Library/API/IoT binding

nanoFramework.System

### Visual Studio version

VS2022  17.5.4

### .NET nanoFramework extension version

2022.3.0.41

### Target name(s)

ESP32

### Firmware version

1.8.1.148

### Device capabilities

n/a

### Description

The UTC offset information is not output when formatting the datetime string, and is treated as a string.

For example:

$"{DateTime.UtcNow:yyyy-MM-ddTHH:mm:ss:z}" produces "2023-05-07T11:48:11:z"

$"{DateTime.UtcNow:yyyy-MM-ddTHH:mm:ss:zz}" produces "2023-05-07T11:48:11:zz"

$"{DateTime.UtcNow:yyyy-MM-ddTHH:mm:ss:zzz}" produces "2023-05-07T11:48:11:zzz"

According to the intellisense, it should be supported to output the offset.

networkfusion, 2023-05-09:
nanoFramework does not support AFIK `local` time offsets, this must be handled in user code.
The reason being that "generally" all IoT devices should be aligned to a globally recognised single source of truth.
If there is a specific piece of documentation that is incorrect, please specify where you found it.
(possibly https://docs.nanoframework.net/content/architecture/date-and-time.html)

josesimoes, 2023-05-09:
Coming from the above explanation, this is by design and there are no plans to add support for local time offsets. 
Closing this one.

closed 2023-05-09 by josesimoes

jasonkaisersmith, 2023-05-09:
I understand that localtime is not supported, but the intellisense states that the timezone offset information is supported as part of the datetime string format.  

Therefore, I would expect that while the timezone is always UTC, it is still included in the output if specified, even if it is always zero.

Also, looking in the code I have found the text below in this file: "https://github.com/nanoframework/CoreLibrary/blob/main/nanoFramework.CoreLibrary/System/Globalization/DateTimeFormat.cs"

    //        "z"     "+0;-0"     timezone offset w/o leading zero      -8
    //        "zz"    "+00;-00"   timezone offset with leading zero     -08
    //        "zzz"   "+00;-00" for hour offset, "00" for minute offset   full timezone offset   -08:00
    //        "zzz*"  "+00;-00" for hour offset, "00" for minute offset   full timezone offset   -08:00

Ellerbach, 2023-05-09:
> Also, looking in the code I have found the text below in this file

Internal comments in the file are most likely copy/paste from .NET. They do not reflect what is available.

> intellisense states that the timezone offset information

Which class states this in which comment? This definitely has to be fixed not to state it.

> Therefore, I would expect that while the timezone is always UTC, it is still included in the output if specified, even if it is always zero.

As explained before, according to ISO, there is no need to have it or have the option available. If you want it, you can easilly manage this on your side. It's about adding, as you've seen in the comment the various elements. 

For nanoFramework, the choice, for simplification has been made not to have any timezone.

jasonkaisersmith, 2023-05-09:
The image shows the intellisense in use:

[image]

Ellerbach, 2023-05-09:
> The image shows the intellisense in use:
> 
> [image]

interesting, that's the intellisense is not coming from the nanoFramework one but from the tool you are using! For date time, you won't see all those details: https://github.com/nanoframework/CoreLibrary/blob/68bb4cb18dd71717b2913bc0fd7684b24ffaf728/nanoFramework.CoreLibrary/System/DateTime.cs#L605

So there is no real way of fixing that, just that the tool is too nice with you and promise you things that we don't have.

jasonkaisersmith, 2023-05-10:
The tool that I am using is VS2022, which is likely what most people are using.  And, I believe, that it generates the information from the class documentation. 

I would suggest, that maybe it is possible that this is being taken from the DateTimeFormat.cs file?

Ellerbach, 2023-05-10:
> I would suggest, that maybe it is possible that this is being taken from the DateTimeFormat.cs file?

It does not there is no intellisense comments there. Then VS is too smart and propose the standard way to users. Most likely related to the usage of `{DateTime.UtcNow:yourform}`. I guess that if you are using `DateTime.UtcNow.ToString("yourformat")`, it will just use the standard comments.