nanoframework/Home #1296: Deployment fails most of time on NUCLEO-F091RC

luciluke, 2023-05-09:
### Tool

Visual Studio extension

### Description

Deployment fails most of time on NUCLEO-F091RC
Tried default empty project with Debug.WriteLine and Blinky examples and sometimes deployment is OK sometimes not. See output. ( STLINK driver is up-to-date)

https://discord.com/channels/478725473862549535/591956247767547910/1105200761459195904

09:38:52.207 [Starting deployment transaction from v2019.12.0.41+076e0df1f2]
09:38:52.209 [Starting debug engine on nanoDevice]
09:38:52.255 [Connecting to debugger engine...OK]
09:38:52.491 [Assemblies to deploy:]
09:38:52.491 [Assembly: NFApp1 v1.0.0.0 (648 bytes)]
09:38:52.590 [Assembly: mscorlib v1.14.3.0 (31876 bytes)]
09:38:52.618 [Assembly: System.Device.Gpio v1.1.28.0 (5868 bytes)]
09:38:52.619 [Assembly: nanoFramework.Runtime.Events v1.11.6.0 (3412 bytes)]
09:38:52.620 [Deploying 4 assemblies to device]
09:38:52.864 [Incrementally deploying assemblies to the device]
09:38:52.928 [Erasing block @ 0x0802B800...]
09:38:52.928 [Deploying 0/41804 bytes.]
09:38:52.932 [Deploying 500/41804 bytes.]
09:38:53.024 [Deploying 1000/41804 bytes.]
09:38:53.024 [Deploying 1500/41804 bytes.]
09:38:53.048 [Deploying 2000/41804 bytes.]
09:38:53.076 [Erasing block @ 0x0802C000...]
... many steps
09:38:56.861 [Deploying 38864/41804 bytes.]
09:38:56.891 [Erasing block @ 0x08035000...]
09:38:56.927 [Deploying 38912/41804 bytes.]
09:38:56.984 [Deploying 39412/41804 bytes.]
09:38:57.067 [Deploying 39912/41804 bytes.]
09:38:57.067 [Deploying 40412/41804 bytes.]
09:38:57.070 [Deploying 40912/41804 bytes.]
09:38:57.099 [Erasing block @ 0x08035800...]
09:38:57.165 [Deploying 40960/41804 bytes.]
09:38:57.166 [Deploying 41460/41804 bytes.]
09:38:57.206 [Assemblies successfully deployed to the device]

!!! nothing changed 10 seconds later !!!

09:39:05.001 [Starting deployment transaction from v2019.12.0.41+076e0df1f2]
09:39:05.003 [Starting debug engine on nanoDevice]
09:39:05.021 [Connecting to debugger engine...OK]
09:39:05.336 [Assemblies to deploy:]
09:39:05.336 [Assembly: NFApp1 v1.0.0.0 (648 bytes)]
09:39:05.524 [Assembly: mscorlib v1.14.3.0 (31876 bytes)]
09:39:05.556 [Assembly: System.Device.Gpio v1.1.28.0 (5868 bytes)]
09:39:05.558 [Assembly: nanoFramework.Runtime.Events v1.11.6.0 (3412 bytes)]
09:39:05.558 [Deploying 4 assemblies to device]
09:39:05.937 [Incrementally deploying assemblies to the device]
09:39:05.937 [Erasing block @ 0x0802B800...]
09:39:05.937 [Deploying 0/41804 bytes.]
09:39:05.937 [Deploying 500/41804 bytes.]
09:39:05.937 [Deploying 1000/41804 bytes.]
09:39:05.937 [Deploying 1500/41804 bytes.]
09:39:10.022 [Error writing 500 bytes to device @ 0x0802BDDC. No reply from nanoDevice.]
09:39:10.022 [*** ERROR deploying assemblies to the device ]
09:39:10.941 [Trying again to deploying assemblies]
09:39:17.009 [ ERROR: failed to get device execution state, aborting deployment ]
09:39:16.948 [ ERROR: deployment failed ***]

Then it can't work 10 times, or it can work again
I would be very grateful for any hint.
-  STLINK driver is up-to-date

### How to reproduce

try to deploy or start a blank project on a NUCLEO-F091RC

### Aditional context

VS version: Microsoft Visual Studio Enterprise 2019 Version 16.11.22

VS extension version: 2019.12.0.41

System Information
HAL build info: nanoCLR running @ ST_NUCLEO64_F091RC
Target: ST_NUCLEO64_F091RC
Platform: STM32F0

Firmware build Info:
Date: Apr 21 2023
Type: MinSizeRel build with ChibiOS v2021.11.3.11
CLR Version: 1.8.1.154
Compiler: GNU ARM GCC v12.2.1

Has nanoBooter: YES
nanoBooter: v1.8.1.154
IFU capable: NO
Has proprietary bootloader: NO

AppDomains:

Assemblies:
NFApp1, 1.0.0.0
mscorlib, 1.14.3.0
System.Device.Gpio, 1.1.28.0
nanoFramework.Runtime.Events, 1.11.6.0

Native Assemblies:
mscorlib v100.5.0.18, checksum 0xCCE8376E
nanoFramework.Runtime.Native v100.0.9.0, checksum 0x109F6F22
nanoFramework.Hardware.Stm32 v100.0.5.1, checksum 0xFE16F347
nanoFramework.ResourceManager v100.0.0.1, checksum 0xDCD7DF4D
nanoFramework.System.Collections v100.0.1.0, checksum 0x2DC2B090
nanoFramework.System.Text v100.0.0.1, checksum 0x8E6EB73D
nanoFramework.Runtime.Events v100.0.8.0, checksum 0x0EAB00C9
EventSink v1.0.0.0, checksum 0xF32F4C3E
System.Device.Gpio v100.1.0.6, checksum 0x097E7BC5
System.Device.I2c v100.0.0.2, checksum 0xFA806D33
System.Device.Pwm v100.1.0.4, checksum 0xABF532C3
System.IO.Ports v100.1.6.1, checksum 0xB798CE30
System.Device.Spi v100.1.2.0, checksum 0x3F6E2A7E

++++++++++++++++++++++++++++++++
++ Memory Map ++
++++++++++++++++++++++++++++++++
Type Start Size
++++++++++++++++++++++++++++++++
RAM 0x20000000 0x00008000
FLASH 0x08000000 0x00040000

+++++++++++++++++++++++++++++++++++++++++++++++++++++++++++++++
++ Flash Sector Map ++
+++++++++++++++++++++++++++++++++++++++++++++++++++++++++++++++
Region Start Blocks Bytes/Block Usage
+++++++++++++++++++++++++++++++++++++++++++++++++++++++++++++++
0 0x08000000 5 0x000800 nanoBooter
1 0x08002800 82 0x000800 nanoCLR
2 0x0802B800 41 0x000800 Deployment

+++++++++++++++++++++++++++++++++++++++++++++++++++
++ Storage Usage Map ++
+++++++++++++++++++++++++++++++++++++++++++++++++++
Start Size (kB) Usage
+++++++++++++++++++++++++++++++++++++++++++++++++++
0x08000000 0x002800 (10kB) nanoBooter
0x08002800 0x029000 (164kB) nanoCLR
0x0802B800 0x014800 (82kB) Deployment

Deployment Map
Empty

subsneer, 2023-05-09:
Also there is a screenshot with timeout caught during nanoff --deploy:
[image]

So deployment has common issue at VS as well as at nanoff

josesimoes, 2024-07-31:
This was fixed in nanoframework/nf-interpreter#2967.
Feel free to reopen if the problem persists.