nanoframework/Home #1298: ESP32 C3 unable to create PWM channel with 50Hz frequency

torbacz, 2023-05-11:
### Target name(s)

ESP32_C3_REV3

### Firmware version

1.8.1.233

### Description

When trying to use servo it's required to use 50Hz PWM channel, yet it fails to create.
[image]

The same code with 5000 frequency works fine, so the pin supports PWM.
[image]

### How to reproduce

```
  const short servoPin = 13;
  nanoFramework.Hardware.Esp32.Configuration.SetPinFunction(servoPin, nanoFramework.Hardware.Esp32.DeviceFunction.PWM1);
  // Set up PWM
  var pwm = PwmChannel.CreateFromPin(servoPin, 50);
```

Ellerbach, 2023-05-12:
@torbacz you mention in the Discord channel that a speed of 50000 works, correct?

Ellerbach, 2023-05-12:
The issue is then somewhere here: https://github.com/nanoframework/nf-interpreter/blob/20c26decce053f1c1b50e1a599928e1b4d1d10b3/targets/ESP32/ESP32_C3/nanoCLR/System.Device.Pwm/sys_dev_pwm_native_System_Device_Pwm_PwmChannel.cpp#L187-L188

PWM implementation for C3 is specific. So most likely something not exactly correct.

Ellerbach, 2023-05-12:
A quick look at the IDF docs shows that for low frequency, some specific settings may need to be applied, that's probably the reason of the issue. See https://docs.espressif.com/projects/esp-idf/en/v4.3/esp32c3/api-reference/peripherals/ledc.html. It seems to fails when trying to find the divider. So more investigations needed.

and indeed, if you have a normal ESP32, you should use it in the meantime.

matsujirushi, 2023-05-21:
Hi @Ellerbach ,

> https://github.com/nanoframework/nf-interpreter/blob/20c26decce053f1c1b50e1a599928e1b4d1d10b3/targets/ESP32/ESP32_C3/nanoCLR/System.Device.Pwm/sys_dev_pwm_native_System_Device_Pwm_PwmChannel.cpp#L187-L188

It looks like the build system doesn't use this file.

**build.ninja:**

```
...
build targets/ESP32/ESP32_C3/CMakeFiles/NF_NativeAssemblies.dir/__/_nanoCLR/System.Device.Pwm/sys_dev_pwm_native_System_Device_Pwm_PwmChannel.cpp.obj: CXX_COMPILER__NF_NativeAssemblies_Debug /workspaces/nf-interpreter/targets/ESP32/_nanoCLR/System.Device.Pwm/sys_dev_pwm_native_System_Device_Pwm_PwmChannel.cpp || cmake_object_order_depends_target_NF_NativeAssemblies
...
```

`/workspaces/nf-interpreter/targets/ESP32/_nanoCLR/System.Device.Pwm/sys_dev_pwm_native_System_Device_Pwm_PwmChannel.cpp`

Is this behavior correct?

matsujirushi, 2023-06-07:
I will try to solve this issue.
Please assign me.

closed 2023-06-09 by josesimoes