nanoframework/Home #1299: Graphics. MakeTransparent

TerryFogg, 2023-05-15:
### Library/API/IoT binding

nanoFramework.Graphics

### Description

Using a colour does not appear to work, while GetPixel() does.

Need to review and understand the problem.

.MakeTransparent(System.Drawing.Color.Magenta);   -- does not work

MakeTransparent(soccerBall.GetPixel(0, 0))  -- works

Ellerbach, 2023-05-15:
You have to adjust the A of the color for that. the A is not handled maybe or transformed or marked as transparent already.
System.Drawing.Color.Magenta = 0xFFFF00FF
For transparent, the A should be 0, so 0x00FF00FF. that's my guess :-)

CoryCharlton, 2023-11-12:
@TerryFogg was your recent Graphics PR related to this?

TerryFogg, 2023-11-12:
This is now fixed. The code used an unusual method to mark a colour as transparent which failed when fixed colours were changed to ARGB format from BGR.  By changing all internal code to ARGB RGB565 and re-coding the transparent detection this is now fixed.

closed 2023-11-12 by TerryFogg