nanoframework/Home #1302: [XIAO_ESP32C3] throws a System.ArgumentException at SpiDevice.Create()

matsujirushi, 2023-05-17:
### Target name(s)

XIAO_ESP32C3

### Description

I am making a sample code for SPI. The device is Seeed Studio XIAO ESP32C3.
I'm trying the code below, but SpiDevice.Create() throws a Sysetm.ArgumentException.

```
Configuration.SetPinFunction(8, DeviceFunction.SPI1_CLOCK); // SCK ... GPIO8
Configuration.SetPinFunction(9, DeviceFunction.SPI1_MISO);  // MISO ... GPIO9
Configuration.SetPinFunction(10, DeviceFunction.SPI1_MOSI); // MOSI ... GPIO10

var settings = new SpiConnectionSettings(1, 20);            // CS ... GPIO20
settings.ClockFrequency = 1_000_000;
settings.Mode = SpiMode.Mode0;
settings.DataFlow = DataFlow.MsbFirst;
settings.DataBitLength = 8;

using var device = SpiDevice.Create(settings);
```

Exception message is here.

```
    ++++ Exception System.ArgumentException - 0xfd000000 (1) ++++
    ++++ Message: 
    ++++ System.Device.Spi.SpiDevice::NativeOpenDevice [IP: 0000] ++++
    ++++ System.Device.Spi.SpiDevice::.ctor [IP: 0019] ++++
    ++++ System.Device.Spi.SpiDevice::Create [IP: 0004] ++++
    ++++ Program::<Main>$ [IP: 0047] ++++
```

### Aditional information

I can solve this problem.
Please assign to me.

Ellerbach, 2023-05-17:
You should use only SPI2 according to the documentation: https://docs.espressif.com/projects/esp-idf/en/latest/esp32c3/api-reference/peripherals/spi_master.html
So do the same with the second bus, should work fine. Otherwise, you should not change the SPI pins.

matsujirushi, 2023-05-18:
FYI
in-progress.
https://github.com/matsujirushi/nf-interpreter/tree/patch/issue1302

alberk8, 2023-05-18:
> FYI in-progress. https://github.com/matsujirushi/nf-interpreter/tree/patch/issue1302

Just PR when done then link it to this issue and that should get it reviewed.

closed 2023-05-18 by josesimoes