nanoframework/Home #1303: Fix Unit Tests in Json library

josesimoes, 2023-05-18:
This issue type is available for Team Members only. If you are not one don't use it and choose another template appropriate to the issue you're trying to report.

### Details about Problem

**nanoFramework area:** nanoFramework.Json

### Description

Following the recent update on the test framework Unit Tests are failing in json library:
https://dev.azure.com/nanoframework/nanoFramework.Json/_build/results?buildId=42342&view=results

Need to clone the PR there run locally and fix them.
https://github.com/nanoframework/nanoFramework.Json/pull/308

Note: this is high priority because its breaking the build.

torbacz, 2023-05-25:
@josesimoes 
I'm not quite sure if it's JSON lib problem. In our project (which is going open source very soon :)) we had problem with serializing/deserializing tests. Tests were failing with exception:
`Standard Error Messages:
     ++++ Exception System.NotSupportedException - 0x00000000 (1) ++++
     ++++ Message: 
     ++++ System.String::GetEnumerator [IP: 0003] ++++
     ++++ nanoFramework.Json.JsonSerializer::SerializeObject [IP: 007d] ++++
     ++++ nanoFramework.Json.JsonSerializer::SerializeIDictionary [IP: 0041] ++++
     ++++ nanoFramework.Json.JsonSerializer::SerializeClass [IP: 0041] ++++
     ++++ nanoFramework.Json.JsonSerializer::SerializeObject [IP: 0089] ++++
     ++++ nanoFramework.Json.JsonConvert::SerializeObject [IP: 0005] ++++
     ++++ MakoIoT.Device.Services.Configuration.ConfigurationService::UpdateConfigSection [IP: 0004] ++++
`

So I've isolated the serialization into normal nano program and deployed on ESP32_REV0. And there was no exception! And today I've downgrade Test lib to 2.1.81 and no more exception on pipeline. I saw that there is a lot of going on in Test nuget, maybe some kind of mistake in there?

josesimoes, 2023-05-26:
Two unrelated things here: the original issue with the unit tests is still there and it's unrelated with what you've described. The other is what you are reporting. Recall that Unit Tests run (by default) with the nanoclr CLI. Altough it shouldn't, there could be a slight difference in the way it's implemented in the CLI build and at the MCU build.

Needs to be investigated. Please raise another issue for that one.
On a second thought it could be related with the way you're accessing the string... I recall seeing that `System.String::GetEnumerator` before... be aware that we don't have support for string enumeration by design, because of space constrains:

https://github.com/nanoframework/CoreLibrary/blob/main/nanoFramework.CoreLibrary/System/String.cs#L31

torbacz, 2023-05-26:
Yep, I saw that it's not implemented. Yet the only thing I'm doing is calling JsonConvert::SerializeObject. So there is something wrong with serializing strings in my opinion. If you look at failing tests in JSON lib, you can find the same stack trace with the same exception.
[image]

I think it's somehow related, something changed in test framework and that's why tests are failing.

josesimoes, 2023-05-26:
Oh!! I missed that detail. Apologies.

Agreed, thi seems to be the same problem.
The only difference that I can think of is the change of the CLR running the tests. It used to be the WIN32 build and it was replaced with the nanoclr CLI. 

Now, I would say that the the root cause is in the serializer.
On quick look, string serialization it's probably (wrongly) going through the `SerializeIEnumerable()` path here

https://github.com/nanoframework/nanoFramework.Json/blob/nfbot/update-dependencies/2a26e3b1-a700-4643-a5bf-504d7661243d/nanoFramework.Json/JsonSerializer.cs#L64-L65

because of `String` implementing `IEnumerable`...

torbacz, 2023-05-26:
No problem :) 

Hmm, maybe that is the case that string implements IEnumerable. Still there is a mystery, why it's working on ESP_REV0... 
Unfortunately, I forgot to bring ESP with me and I'm out for weekend, I can try to change implementation and try to run tests on nanoclr. Let's hope this time it will worki 😄

josesimoes, 2023-05-26:
Nice! If you can look into this, I would really appreciate it.

As a side comment: there should be no difference on the string not implementing GetEnumerator among builds. So, as you'«ve pointed, this doesn't explain how the ESP build passes. Unless you're running it with the "old" WIN32 test runner.

torbacz, 2023-05-27:
There is a problem in Hashtable implementation. String type should never go into IEnumerable, because there is custom mapper for string. They are stored in hashtable and resolved by type 
[image]

After some investigation, changing hashtable key from "Type" to "string" (full type name) it seems to work fine.
Potential fix, haven't tests performance https://github.com/nanoframework/nanoFramework.Json/pull/309

closed 2023-05-28 by josesimoes