nanoframework/Home #1304: Fix Unit Tests in Azure.Devices library

josesimoes, 2023-05-18:
### Details about Problem

**nanoFramework area:** nanoFramework.Azure.Devices

### Description

Following the recent update on the test framework Unit Tests are failing in nanoFramework.Azure.Devices library:
https://dev.azure.com/nanoframework/Azure.Devices/_build/results?buildId=42349&view=logs&j=12f1170f-54f2-53f3-20dd-22fc7dff55f9&t=7bfb7e0f-549d-5929-c603-6e8046613b9e

Need to clone the PR there run locally and fix them.
https://github.com/nanoframework/nanoFramework.Azure.Devices/pull/291

Note: this is high priority because its breaking the build.

torbacz, 2023-05-28:
@josesimoes Fixed already?

josesimoes, 2023-05-28:
Yes. Unit tess all a green again! 👍

josesimoes, 2023-05-28:
Nope. Still open.

torbacz, 2023-05-29:
But new pipelines are working, with latest version of test framework
[image]

josesimoes, 2023-05-29:
Yes, that got fixed too. I was under the impression that it wasn't... apologies

josesimoes, 2023-05-29:
Closed by nanoframework/nanoFramework.Azure.Devices#292

closed 2023-05-29 by josesimoes