nanoframework/Home #1306: Large (>80K) byte[] array definition causes A2000000 error on deployment.

skigrinder, 2023-05-26:
### Target name(s)

XIAO_ESP32C3  ESP32_REV0

### Firmware version

v1.8.1.275

### Was working before? On which version?

N/A

### Device capabilities

06:17:05.522 [NanoDevices: valid device XIAO_ESP32C3 @ COM10]
06:17:05.523 [NanoDevices: Serial device enumeration completed. Found 1 devices]

System Information
HAL build info: nanoCLR running @ ESP32_C3 built with ESP-IDF e8bdaf9
  Target:   XIAO_ESP32C3
  Platform: ESP32

Firmware build Info:
  Date:        May 19 2023
  Type:        MinSizeRel build, chip rev. >= 3, without support for PSRAM
  CLR Version: 1.8.1.264
  Compiler:    GNU ARM GCC v8.4.0

OEM Product codes (vendor, model, SKU): 0, 0, 0

Serial Numbers (module, system):
  00000000000000000000000000000000
  0000000000000000

Target capabilities:
  Has nanoBooter: NO
  IFU capable: NO
  Has proprietary bootloader: YES

AppDomains:

Assemblies:
  Sample01, 1.0.0.0
  System.Device.I2c, 1.1.11.0
  nanoFramework.Runtime.Events, 1.11.6.0
  System.Device.Gpio, 1.1.28.0
  nanoFramework.Hardware.Esp32, 1.6.3.0
  nanoFramework.Logging, 1.1.63.0
  mscorlib, 1.14.3.0
  Mort.VL53L5CX, 1.0.0.0
  Mort.VL53L5CXfw1, 1.0.0.0
  VL53L5CXfw2, 1.0.0.0

Native Assemblies:
  mscorlib v100.5.0.18, checksum 0xCCE8376E
  nanoFramework.Runtime.Native v100.0.9.0, checksum 0x109F6F22
  nanoFramework.Hardware.Esp32 v100.0.9.0, checksum 0xC0E1540A
  nanoFramework.Networking.Sntp v100.0.4.4, checksum 0xE2D9BDED
  nanoFramework.ResourceManager v100.0.0.1, checksum 0xDCD7DF4D
  nanoFramework.System.Collections v100.0.1.0, checksum 0x2DC2B090
  nanoFramework.System.Text v100.0.0.1, checksum 0x8E6EB73D
  nanoFramework.Runtime.Events v100.0.8.0, checksum 0x0EAB00C9
  EventSink v1.0.0.0, checksum 0xF32F4C3E
  System.IO.FileSystem v1.0.0.0, checksum 0x3AB74021
  System.Math v100.0.5.4, checksum 0x46092CB1
  System.Net v100.1.5.0, checksum 0x5BAB8CB3
  System.Device.Adc v100.0.0.0, checksum 0xE5B80F0B
  System.Device.Gpio v100.1.0.6, checksum 0x097E7BC5
  System.Device.I2c v100.0.0.2, checksum 0xFA806D33
  System.Device.Pwm v100.1.0.4, checksum 0xABF532C3
  System.IO.Ports v100.1.6.1, checksum 0xB798CE30
  System.Device.Spi v100.1.2.0, checksum 0x3F6E2A7E
  System.Runtime.Serialization v100.0.0.0, checksum 0x0A066871
  System.Device.Wifi v100.0.6.4, checksum 0x00A058C6
  Windows.Storage v100.0.2.0, checksum 0x954A4192

++++++++++++++++++++++++++++++++
++        Memory Map          ++
++++++++++++++++++++++++++++++++
  Type     Start       Size
++++++++++++++++++++++++++++++++
       0x4b50464e  0x00315654
       0x00000000  0x00000001

+++++++++++++++++++++++++++++++++++++++++++++++++++++++++++++++
++                   Flash Sector Map                        ++
+++++++++++++++++++++++++++++++++++++++++++++++++++++++++++++++
  Region     Start      Blocks   Bytes/Block    Usage
+++++++++++++++++++++++++++++++++++++++++++++++++++++++++++++++
      0    0x00010000       1      0x1A0000     nanoCLR
      1    0x001B0000       1      0x1F0000     Deployment
      2    0x003C0000       1      0x040000     Configuration

+++++++++++++++++++++++++++++++++++++++++++++++++++
++              Storage Usage Map                ++
+++++++++++++++++++++++++++++++++++++++++++++++++++
  Start        Size (kB)           Usage
+++++++++++++++++++++++++++++++++++++++++++++++++++
  0x003C0000    0x040000 (256kB)    Configuration
  0x00010000    0x1A0000 (1664kB)   nanoCLR
  0x001B0000    0x1F0000 (1984kB)   Deployment

Deployment Map
Empty

### Description

If the following byte[] definition contains more than 80K bytes, deploying the solution will cause an A2000000 error:
The following code segment will cause an A2000000 error on deployment (or debug):

    public static class Firmware1 {
        public static readonly byte[] foobar = new byte[] {
             0xe0, 0x00, 0x03, 0x08,
             0xe0, 0x00, 0x0a, 0xc8,
             0xe0, 0x00, 0x05, 0x08,
             0xe0, 0x60, 0x37, 0xc8,
             0xe0, 0x00, 0x0a, 0x88,
             0xe0, 0x00, 0x0a, 0x88,
             0xe0, 0x00, 0x0a, 0x88,
             .......  for more than 80K bytes
         }
     }

### How to reproduce

Please see Description section

### Expected behaviour

If a large byte[] is split into multiple, smaller (i.e. <60K) byte[] definitions, the solution will deploy correctly.
Experimentation showed that the breaking point is somewhere between 60K and 80K.
However, testing to determine the absolute upper limit for a given target (i.e. the XIAO ESP32C3) was not done.

Expected behavior:
If two 40K byte[] definitions deploy correctly, then a single 80K byte[] definition should also deploy correctly.

### Screenshots

Please see Description section

### Aditional information

Please see Description section

Ellerbach, 2023-05-26:
@skigrinder thanks for the detailed issue. Without PSRAM, it seems normal that you can load a 80K byte array. It is normal as well you can 2 of 40K approximately. My understandind is that it is related to a limitation in how the memory is manaed on the IDF underlying with non continuous amount of memory.

With PSRAM, then the story is very different and you should be able to have your 80K without any issue.

@josesimoes to confirm or infirm.

torbacz, 2023-05-26:
Isn't that expected? Array is stored in order in memory, there has to be block with amount you need. 2 x 40K needs two block of at least 40kB and one 80kB needs at least 80kB. You probably run into situation when, you've got for example 50kB and 60kB block available, so two 40kB fits, but one 80kB can't.

josesimoes, 2023-09-01:
I've just tested this on a ESP32_C3 with fw ESP32_C3_REV3-1.8.1.491.

```csharp
Console.WriteLine("Hello World!");

var buffer = new byte[1024 * 80];

Console.WriteLine($"BUffer has: {buffer.Length }");
```

output was:

```text
Hello World!
BUffer has: 81920
```

skigrinder, 2023-09-02:
Thanks Jose'!
I think @torbacz was correct.  Breaking the 80K data into 30K chunks got my application running.
At your suggestion, I switched from referencing the .dll to referencing the project and the 0xA2000000 went away.

josesimoes, 2023-09-02:
Ok... What was the project that you've switched to reference?

josesimoes, 2023-09-04:
OK. I guess this can be closed now?

skigrinder, 2023-09-04:
I agree.

closed 2023-09-04 by josesimoes