nanoframework/Home #1307: ESP32 latest Firmware in boot loop

alberk8, 2023-05-27:
### Target name(s)

ESP32_REV3, ESP32_PSRAM_REV3

### Firmware version

1.8.1.275

### Was working before? On which version?

1.8.1.264

### Description

It is in a boot loop

### How to reproduce

1) Flash with the latest firmware
2) The ESP32 does not appear on Visual studio 2022 Device Manager
3) Putty in  with baud rate 115200

### Expected behaviour

There is no boot loop.

### Screenshots

[image]

josesimoes, 2023-05-28:
This is with REV3 only, correct?

alberk8, 2023-05-29:
Tested on an REV0 hardware.  The ESP32_REV0 is doing boot loop where as the **ESP32_PSRAM_REV0 is ok**.