nanoframework/Home #1316: Update UI samples to new API

josesimoes, 2023-06-02:
### Details about Problem

**nanoFramework area:** Samples

### Description

Build of samples is failing for the UI/Graphics related ones: https://github.com/nanoframework/Samples/pull/304
This is because of the recent changes in UI APIs.

Trivial work required: build locally and fix the code.

Medium priority from the fact that this is breaking the build on the pipeline and updating all samples is failing.

networkfusion, 2025-03-07:
@josesimoes @Ellerbach the linked PR is fixed. Is the issue resolved?

josesimoes, 2025-03-07:
Most likely yes... the samples have been updated smoothly, so I suppose everything is OK now. If not, this can be revisited.

closed 2025-03-07 by josesimoes