nanoframework/Home #1317: System.Net.Sockets.SocketException - CLR_E_FAIL (1):  During https request

jasonkaisersmith, 2023-06-06:
### Library/API/IoT binding

nanoFramework.Networking

### Visual Studio version

VS2022  17.5.6

### .NET nanoFramework extension version

2022.3.0.41

### Target name(s)

ESP32

### Firmware version

1.8.1.307

### Description

When executing https request then the process is failing with the following:

 ++++ Exception System.Net.Sockets.SocketException - CLR_E_FAIL (1) ++++
    ++++ Message: 
    ++++ System.Net.Security.SslNative::SecureConnect [IP: 0000] ++++
    ++++ System.Net.Security.SslStream::Authenticate [IP: 0054] ++++
    ++++ System.Net.Security.SslStream::AuthenticateAsClient [IP: 000a] ++++
    ++++ System.Net.HttpWebRequest::EstablishConnection [IP: 027c] ++++
    ++++ System.Net.HttpWebRequest::SubmitRequest [IP: 001a] ++++
    ++++ System.Net.HttpWebRequest::GetResponse [IP: 000c] ++++

This fails both in my code, **and the official samples** "HttpWebRequest", therefore I have decided to move my prior discussion and create this bug report instead.

When executing the code, if I use the debugger and check the HttpClient after set-up but prior to execution of the request then I have noticed that all the SSLProtocols value is set to Tls12, if you view the underlying HttpClientHandler then this has "None" for the same value.

See the attached screenshot where I have highlighted them both.

### Screenshots

[image]

### Sample project or code

private const string myCACertificate = @"-----BEGIN CERTIFICATE-----   text of my cert removed for brevity -----END CERTIFICATE-----";

     private static  HttpClient _httpClient = new HttpClient()
     {
            HttpsAuthentCert = new X509Certificate(myCACertificate),
            SslProtocols = System.Net.Security.SslProtocols.Tls12
     };

    string json = JsonConvert.SerializeObject(new MyObject());
    StringContent strcontent = new StringContent(json, Encoding.UTF8, "application/json");

    string url = $"https://myserver:8788/api/path";

    var response = _httpClient.Post(url, strcontent);

Ellerbach, 2023-06-07:
Is it the same as this one? https://github.com/nanoframework/Home/issues/1319
I have multiple devices working properly with certificates without any issue. I can't test if any regression for the next few days.

networkfusion, 2023-12-27:
Please provide feedback, otherwise the issue will be closed.

jasonkaisersmith, 2023-12-29:
Sorry, but I moved away from using the framework as this was a show stopper, plus I had too many stability issues in general.  So I am no longer able to verify it.

FYI:  On Mongoose O/S it worked without any issue with the same certificate.

networkfusion, 2023-12-29:
Closing as likely working. Feel free to re-open if necessary.

closed 2023-12-29 by networkfusion

belem2050, 2025-02-12:
@Ellerbach @networkfusion 
``` ++++ Exception System.Net.Sockets.SocketException - CLR_E_FAIL (6) ++++
 ++++ Message: 
 ++++ System.Net.Security.SslNative::SecureAccept [IP: 0000] ++++
 ++++ System.Net.Security.SslStream::Authenticate [IP: 0032] ++++
 ++++ System.Net.Security.SslStream::AuthenticateAsServer [IP: 0016] ++++
 ++++ System.Net.HttpListener::AcceptThreadFunc [IP: 009a] ++++
 ++++ Exception System.Net.Sockets.SocketException - CLR_E_FAIL (6) ++++
 ++++ Message: 
 ++++ System.Net.Security.SslNative::SecureAccept [IP: 0000] ++++
 ++++ System.Net.Security.SslStream::Authenticate [IP: 0032] ++++
 ++++ System.Net.Security.SslStream::AuthenticateAsServer [IP: 0016] ++++
 ++++ System.Net.HttpListener::AcceptThreadFunc [IP: 009a] ++++
 ++++ Exception System.Net.Sockets.SocketException - CLR_E_FAIL (6) ++++
 ++++ Message: 
 ++++ System.Net.Security.SslNative::SecureAccept [IP: 0000] ++++
 ++++ System.Net.Security.SslStream::Authenticate [IP: 0032] ++++
 ++++ System.Net.Security.SslStream::AuthenticateAsServer [IP: 0016] ++++
 ++++ System.Net.HttpListener::AcceptThreadFunc [IP: 009a] ++++
 ++++ Exception System.Net.Sockets.SocketException - CLR_E_FAIL (6) ++++
 ++++ Message: 
 ++++ System.Net.Security.SslNative::SecureAccept [IP: 0000] ++++
 ++++ System.Net.Security.SslStream::Authenticate [IP: 0032] ++++
 ++++ System.Net.Security.SslStream::AuthenticateAsServer [IP: 0016] ++++
 ++++ System.Net.HttpListener::AcceptThreadFunc [IP: 009a] ++++
```
This is what I am having from wireshark:

[image]

It connects once and then, infinitely tries to establish a connection.
I have been implementing a server using nanoframework webserver and I got everything working fine with HTTP and some times with HTTPS. But unfortunately, I am constantly having a ``ERR_CONNECTION_RESET```while trying to access from the browser. I've also tried some code from HTTP client in C sharp Native but I couldn't get it to worked since it raising the the kind of same issue. 
I t looks like coming from some SSL stuff but couldn't fine that could help. So I am seeking for help here and anyone has an idea what it could be, I'm willing to hear. Thanks in advance.