nanoframework/Home #1318: Sample HttpWebRequest does not work.  Fails to get resource

jasonkaisersmith, 2023-06-06:
### Library/API/IoT binding

HttpWebRequestSample

### Visual Studio version

VS2022  17.5.6

### .NET nanoFramework extension version

2022.3.0.41

### Target name(s)

ESP32

### Firmware version

1.8.1.307

### Description

The official same "HttpWebRequestSample" does not work.

It fails at this line:
X509Certificate rootCACert = new X509Certificate(Resources.GetBytes(Resources.BinaryResources.DigiCertGlobalRootCA));

It is unable to retrieve the certificate "DigiCertGlobalRootCA.crt" from the resources in the project during runtime.

### How to reproduce

Run the sample.

Ellerbach, 2023-06-07:
I can't try it for the next few days. But can you try to remove the resource file fully? Then add it again (beware there is a specific nanoFramework resource file) and see if that fixes things?

CoryCharlton, 2023-10-04:
There seems to be an issue with positive resource ids and fonts. That might be related to this issue :shrug:

josesimoes, 2023-10-04:
I've just cloned, restore nugets, rebuild (as usual) and have it working:

[image]

Request goes through and I receive the HTML page...

[image]

Soo.... not sure what could be wrong... need more details.

networkfusion, 2023-12-27:
Closing as no feedback. Please feel free to reopen if required.

closed 2023-12-27 by networkfusion