nanoframework/Home #1319: Add support for ServerCertificateCustomValidationCallback

jasonkaisersmith, 2023-06-06:
### Description

When developing web apps for internal use, it is common to use self-signed certificates, or certificates not verified by a recognised CA authority.

In .net it is possible to use "ServerCertificateCustomValidationCallback" to catch certificate errors thrown by the framework and then handle this so that the user can suppress the problem as they wish (at their own risk of course!)

### How to solve the problem

If something similar to this would be possible then that would be good.

var handler = new HttpClientHandler(); 
handler.ServerCertificateCustomValidationCallback += CertificateSelfValidation;

var httpClient = new HttpClien(handler);

private  bool CertificateSelfValidation(object sender, X509Certificate certificate, X509Chain chain, SslPolicyErrors sslPolicyErrors)
{
	//Example take from https://learn.microsoft.com/en-us/dotnet/fundamentals/code-analysis/quality-rules/ca5359
	 if (sslPolicyErrors == SslPolicyErrors.None)
        {
            return true;
        }

    // For HTTPS requests to this specific host, we expect this specific certificate.
    // In practice, you'd want this to be configurable and allow for multiple certificates per host, to enable
    // seamless certificate rotations.
    return sender is HttpWebRequest httpWebRequest
                && httpWebRequest.RequestUri.Host == "localhost"
                && certificate is X509Certificate2 x509Certificate2
                && x509Certificate2.Thumbprint == "AAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAA"
                && sslPolicyErrors == SslPolicyErrors.RemoteCertificateChainErrors;
}

josesimoes, 2023-06-06:
What you are suggesting is not possible (or at least easily doable) because the TLS negotiation it's handled by the native code and it's complex to "interrupt" it, do the thing in the validation callback and resume the process. That's the reason why it's not available.

The [`SslStream`](https://docs.nanoframework.net/api/System.Net.Security.SslStream.html#System_Net_Security_SslStream_SslVerification) exposes a property that would allow you to acomplish what you're asking here. That's [`SslVerification`](https://docs.nanoframework.net/api/System.Net.Security.SslStream.html#System_Net_Security_SslStream_SslVerification).
Unfortunatly that one is not exposed in `HttpClient`...

Can I interest you on submitting a PR to the HttpClient library with that improvement? 😅 
(Happy to help and/or provide any guidance that you need)

Alex-111, 2023-07-30:
@josesimoes does sslverification really control the verification of the server certificate, or is it just a feature to skip checking the client certificate?

josesimoes, 2023-07-31:
@Alex-111 that skips validanting the SERVER certificate. It has nothing to do with client certificates.

closed 2023-10-04 by josesimoes

josesimoes, 2023-10-04:
`HttpClient` now exposes `SslVerification` field allowing to bypass server certificate validation. 
Feedback welcome!