nanoframework/Home #1324: API docs aren't linking to the source code file

josesimoes, 2023-06-16:
### Details about Problem

**nanoFramework area:** ( docs )

### Description

Standard docfx generated pages show a link to "Improve this document" and for APIs the methods link to the source code file in the respective github repo.
Like this:

[image]

This used to be there. As it is today it's not. Something in the docfx config or the parsing that occurs in the pipeline must be preventing this from being added to the docs.

This is a very convinient feature and we should have it back.

josesimoes, 2023-06-16:
@Ellerbach maybe you have some thoughts about this... 😉

Ellerbach, 2023-06-18:
> @Ellerbach maybe you have some thoughts about this... 😉

Yes, indeed, the "origin" repo is used as the source, so nanoframework.github.io. Because we clone all the rest, assemble elements from everywhere, the source is not adjusted. I haven't checked in details if there is a way to adjust this. Most probably, we would have to deactivate this feature.

josesimoes, 2023-06-18:
This used to be there. I recall having used it in the past to navigate to the source code file in the repo. I'm guessing something is missing for docfx to be able to link to it.

closed 2023-08-04 by Ellerbach