nanoframework/Home #1325: [Samples]  Add  DigiCert Global Root G2 root CA certificate

matsujirushi, 2023-06-21:
### Description

The Azure IoT Hub and Azure IoT Hub DPS will migrate from the Baltimore CyberTrust Root to the DigiCert Global G2 Root starting February 15, 2023, completing on February 15, 2024.

https://techcommunity.microsoft.com/t5/internet-of-things-blog/azure-iot-tls-critical-changes-are-almost-here-and-why-you/ba-p/2393169

Some sample code is just baltimore's certificate.

### How to solve the problem

Add  DigiCert Global Root G2 root CA certificate.

Ellerbach, 2023-06-21:
The sample has both certificates, you can just add the one you need.
If you're willing to give it a try, you can even add both and adjust the connection to check the first one first, if not the second one.

Ellerbach, 2023-06-21:
Fair enough. Feel free to adjust the samples and raise a PR. We ❤️ contributions!

matsujirushi, 2023-06-22:
Please assign me this issue. 👍
I will create PRs.

closed 2023-08-04 by Ellerbach