nanoframework/Home #1332: GpioPin.ValueChanged unstable

matsujirushi, 2023-07-14:
### Library/API/IoT binding

nanoFramework.System.Device.Gpio

### Visual Studio version

VS2022

### .NET nanoFramework extension version

2022.3.0.41

### Target name(s)

ESP32_REV3

### Firmware version

1.8.1.338

### Device capabilities

System Information
HAL build info: nanoCLR running @ ESP32 built with ESP-IDF e8bdaf9
  Target:   ESP32_REV3
  Platform: ESP32

Firmware build Info:
  Date:        Jun 23 2023
  Type:        MinSizeRel build, chip rev. 3, without support for PSRAM
  CLR Version: 1.8.1.338
  Compiler:    GNU ARM GCC v8.4.0

OEM Product codes (vendor, model, SKU): 0, 0, 0

Serial Numbers (module, system):
  00000000000000000000000000000000
  0000000000000000

Target capabilities:
  Has nanoBooter: NO
  IFU capable: NO
  Has proprietary bootloader: YES

AppDomains:

Assemblies:
  GpioPinValueChanged, 1.0.0.0
  nanoFramework.Runtime.Events, 1.11.6.0
  mscorlib, 1.14.3.0
  System.Device.Gpio, 1.1.28.0

Native Assemblies:
  mscorlib v100.5.0.18, checksum 0xCCE8376E
  nanoFramework.Runtime.Native v100.0.9.0, checksum 0x109F6F22
  nanoFramework.Hardware.Esp32 v100.0.9.0, checksum 0xC0E1540A
  nanoFramework.Hardware.Esp32.Rmt v100.0.4.0, checksum 0x608C5658
  nanoFramework.Device.OneWire v100.0.4.0, checksum 0xB95C43B4
  nanoFramework.Networking.Sntp v100.0.4.4, checksum 0xE2D9BDED
  nanoFramework.ResourceManager v100.0.0.1, checksum 0xDCD7DF4D
  nanoFramework.System.Collections v100.0.1.0, checksum 0x2DC2B090
  nanoFramework.System.Text v100.0.0.1, checksum 0x8E6EB73D
  nanoFramework.Runtime.Events v100.0.8.0, checksum 0x0EAB00C9
  EventSink v1.0.0.0, checksum 0xF32F4C3E
  System.IO.FileSystem v1.0.0.0, checksum 0x3AB74021
  System.Math v100.0.5.4, checksum 0x46092CB1
  System.Net v100.1.5.0, checksum 0x5BAB8CB3
  System.Device.Adc v100.0.0.0, checksum 0xE5B80F0B
  System.Device.Dac v100.0.0.6, checksum 0x02B3E860
  System.Device.Gpio v100.1.0.6, checksum 0x097E7BC5
  System.Device.I2c v100.0.0.2, checksum 0xFA806D33
  System.Device.I2s v100.0.0.1, checksum 0x478490FE
  System.Device.Pwm v100.1.0.4, checksum 0xABF532C3
  System.IO.Ports v100.1.6.1, checksum 0xB798CE30
  System.Device.Spi v100.1.2.0, checksum 0x3F6E2A7E
  System.Runtime.Serialization v100.0.0.0, checksum 0x0A066871
  System.Device.Wifi v100.0.6.4, checksum 0x00A058C6
  Windows.Storage v100.0.2.0, checksum 0x954A4192

### Description

Sometimes Rising and Falling do not occur alternately in GpioPin.ValueChanged.

**Sample code:**
```csharp
using System;
using System.Device.Gpio;
using System.Diagnostics;
using System.Threading;

namespace GpioPinValueChanged
{
    public class Program
    {
        public static void Main()
        {
            using var gpio = new GpioController();
            using var button = gpio.OpenPin(0, PinMode.InputPullUp);
            button.DebounceTimeout = TimeSpan.FromMilliseconds(1000);
            button.ValueChanged += Button_ValueChanged;

            Thread.Sleep(Timeout.Infinite);
        }

        private static void Button_ValueChanged(object sender, PinValueChangedEventArgs e)
        {
            var now = DateTime.UtcNow;
            Debug.WriteLine($"{DateTime.UtcNow.TimeOfDay.TotalSeconds:f3} {e.ChangeType.ToString()}");
        }
    }
}
```

**Output:**
```
...
551.490 2
555.219 1
557.040 2
573.329 2
574.479 1
584.189 2
...
```

josesimoes, 2023-07-14:
That may be happening because the debounce is filtering the change? or there is too much rigging on the signal?

matsujirushi, 2023-07-17:
I think there is a possibility that debounce is affecting it.
Just a possibility, I don't know where the problem is.

josesimoes, 2023-09-01:
@matsujirushi any more feedback on this? Have you manged to test against the debounce being the cause for the instability your complaining about?

matsujirushi, 2023-09-03:
@josesimoes Thank you for message.
I'll close it because it will take some time to confirm. Reopen if necessary.

closed 2023-09-03 by matsujirushi

mikeobrien, 2023-10-08:
Nearly identical setup as the OP and I'm running into the same thing. Doing this:

```
var button = GpioController.OpenPin(2, PinMode.InputPullUp);

button.DebounceTimeout = TimeSpan.FromMilliseconds(20);

button.ValueChanged += (s, e) =>
{
    Debug.WriteLine($"{DateTime.UtcNow.ToString("mm:ss.fff")} {e.ChangeType}");
};
```
And I just produced these results repeatedly pressing and releasing:
```
49:37.806 Falling
49:37.835 Rising
49:40.615 Falling
49:41.115 Rising
49:42.425 Falling
49:42.915 Rising
49:44.615 Falling
49:44.824 Rising
49:47.215 Rising
49:56.635 Rising
49:59.455 Rising
50:01.505 Rising
50:03.575 Rising
50:05.725 Rising
50:07.295 Falling
50:07.776 Rising
50:10.025 Rising
```
Tried higher debounce times but it eventually ends up making it miss more and more presses. I did try this with two different switches (a very bouncy one and a nearly no bounce one) and that does make a difference but still had the problem. Tried *not* setting the debounce timeout and handling it myself and was STILL getting multiple bounce events far apart, as in > 100 ms so couldn't use a reasonable timeout (and not a reentrance issue). Lastly, tried `GpioButton` with even worse results (looks like it implements it's own debouncing). I have a feeling this isn't about debouncing per se but an issue with events that the bouncing brings to light. 

Now if I implement debouncing in a tight loop with a 20 ms debounce it works flawlessly (no matter what rate I press or what button I use). 
```
var gpioController = new GpioController();
var button = gpioController.OpenPin(2, PinMode.InputPullUp);
var debounceStart = DateTime.UtcNow;
var inDebounce = false;
var lastEvent = PinValue.High;

while (true)
{
    var now = DateTime.UtcNow;

    var state = button.Read();

    if ((inDebounce && (DateTime.UtcNow - debounceStart).TotalMilliseconds <= 10) ||
        state == lastEvent)
    {
        continue;
    }

    inDebounce = true;
    debounceStart = DateTime.UtcNow;
    lastEvent = state;

    Debug.WriteLine($"{now.ToString("mm:ss.fff")} {state}");
}
```

```
19:16.150 Low
19:16.281 High
19:17.943 Low
19:18.088 High
19:18.880 Low
19:18.971 High
19:19.039 Low
19:19.134 High
19:20.132 Low
19:20.465 High
19:21.385 Low
19:21.678 High
19:23.708 Low
19:24.008 High
19:27.087 Low
19:27.434 High
19:38.164 Low
19:38.300 High
19:40.087 Low
19:42.522 High
```

mikeobrien, 2023-10-08:
Took the tight loop one step further and completely removed the debounce code and still works flawlessly. This switch had quite a bit of noise on my Teensy 4.1 so had to debounce it on there. Completely baffled by this... TL;DR loop works perfect with no debouncing, events act like they are bouncing and debouncing timeout doesn't fix it.

```
var gpioController = new GpioController();
var button = gpioController.OpenPin(2, PinMode.InputPullUp);
var lastEvent = PinValue.High;

while (true)
{
    var state = button.Read();

    if (state == lastEvent) continue;
    
    lastEvent = state;

    Debug.WriteLine($"{DateTime.UtcNow.ToString("mm:ss.fff")} {state}");
}
```

```
21:52.153 Low
21:52.289 High
21:52.867 Low
21:52.957 High
21:54.222 Low
21:54.461 High
21:54.587 Low
21:54.692 High
21:55.202 Low
21:57.273 High
21:57.666 Low
21:57.919 High
21:58.515 Low
21:58.643 High
21:58.731 Low
21:58.848 High
21:59.153 Low
21:59.235 High
21:59.416 Low
21:59.531 High
21:59.619 Low
21:59.712 High
21:59.885 Low
22:00.005 High
22:00.337 Low
22:00.430 High
22:00.688 Low
22:00.840 High
22:01.110 Low
22:01.641 High
```

josesimoes, 2023-10-09:
@mikeobrien on what platform are you running this?

mikeobrien, 2023-10-09:
[ESP32-S3-DEVKITC-1-N8R8-ND](https://www.digikey.com/en/products/detail/espressif-systems/ESP32-S3-DEVKITC-1-N8R8/15295894)

```
System Information
HAL build info: nanoCLR running @ ESP32_S3 built with ESP-IDF v4.4.5
  Target:   ESP32_S3
  Platform: ESP32

Firmware build Info:
  Date:        Sep 27 2023
  Type:        MinSizeRel build, support for PSRAM
  CLR Version: 1.8.1.546
  Compiler:    GNU ARM GCC v8.4.0

Target capabilities:
  Has nanoBooter: NO
  IFU capable: NO
  Has proprietary bootloader: YES

Assemblies:
  AftOverhead, 1.0.0.0
  nanoFramework.Runtime.Events, 1.11.6.0
  System.Device.Gpio, 1.1.28.0
  System.Net, 1.10.52.0
  System.Threading, 1.1.19.33722
  nanoFramework.Graphics.Core, 1.1.37.0
  System.Device.Wifi, 1.5.54.0
  nanoFramework.System.Text, 1.2.37.0
  System.IO.Streams, 1.1.38.0
  Iot.Device.Button, 1.2.0.0
  nanoFramework.System.Collections, 1.5.18.0
  System.IO.FileSystem, 1.1.23.0
  Iot.Device.Ws28xx.Esp32, 1.2.0.0
  mscorlib, 1.14.3.0
  nanoFramework.Hardware.Esp32, 1.6.8.0
  nanoFramework.Hardware.Esp32.Rmt, 2.0.1.0
  Common, 1.0.0.0
  nanoFramework.Runtime.Native, 1.6.6.0
  nanoFramework.Json, 2.2.99.0
  nanoFramework.DependencyInjection, 1.0.35.0
  System.Net.Http, 1.5.102.0
  Iot.Device.DhcpServer, 1.2.0.0

Native Assemblies:
  mscorlib v100.5.0.18, checksum 0xCCE8376E
  nanoFramework.Runtime.Native v100.0.9.0, checksum 0x109F6F22
  nanoFramework.Hardware.Esp32 v100.0.10.0, checksum 0x6A20A689
  nanoFramework.Hardware.Esp32.Rmt v100.0.4.0, checksum 0x608C5658
  nanoFramework.Networking.Sntp v100.0.4.4, checksum 0xE2D9BDED
  nanoFramework.ResourceManager v100.0.0.1, checksum 0xDCD7DF4D
  nanoFramework.System.Collections v100.0.1.0, checksum 0x2DC2B090
  nanoFramework.System.Text v100.0.0.1, checksum 0x8E6EB73D
  nanoFramework.Runtime.Events v100.0.8.0, checksum 0x0EAB00C9
  EventSink v1.0.0.0, checksum 0xF32F4C3E
  System.IO.FileSystem v1.0.0.0, checksum 0x3AB74021
  System.Math v100.0.5.4, checksum 0x46092CB1
  System.Net v100.1.5.0, checksum 0x5BAB8CB3
  System.Device.Adc v100.0.0.0, checksum 0xE5B80F0B
  System.Device.Gpio v100.1.0.6, checksum 0x097E7BC5
  System.Device.I2c v100.0.0.2, checksum 0xFA806D33
  System.Device.I2s v100.0.0.1, checksum 0x478490FE
  System.Device.Pwm v100.1.0.4, checksum 0xABF532C3
  System.IO.Ports v100.1.6.1, checksum 0xB798CE30
  System.Device.Spi v100.1.2.0, checksum 0x3F6E2A7E
  System.Runtime.Serialization v100.0.0.0, checksum 0x0A066871
  System.Device.Wifi v100.0.6.4, checksum 0x00A058C6
  Windows.Storage v100.0.2.0, checksum 0x954A4192
```

josesimoes, 2023-10-10:
I've just check the code briefly. I don't think there much we can do to improve it as it is...
Know that debouncing on a C# loop it's resource intensive and that any `Thread.Sleep()` under 20ms is going to hit the execution performance as it will be constantly executing.