nanoframework/Home #1336: ESP32-S3 firmware cannot recognize the PSRAM

sd3326852, 2023-07-28:
### Target name(s)

ESP32-S3

### Firmware version

1.8.1.384

### Was working before? On which version?

No

### Device capabilities

System Information
HAL build info: nanoCLR running @ ESP32_S3 built with ESP-IDF ac5d805
  Target:   ESP32_S3_BLE
  Platform: ESP32

Firmware build Info:
  Date:        Jul 27 2023
  Type:        MinSizeRel build, support for PSRAM, support for BLE
  CLR Version: 1.8.1.384
  Compiler:    GNU ARM GCC v8.4.0

OEM Product codes (vendor, model, SKU): 0, 0, 0

Serial Numbers (module, system):
  00000000000000000000000000000000
  0000000000000000

Target capabilities:
  Has nanoBooter: NO
  IFU capable: NO
  Has proprietary bootloader: YES

AppDomains:

Assemblies:
  NFToy, 1.0.0.0
  nanoFramework.Runtime.Events, 1.11.6.0
  Iot.Device.Ws28xx.Esp32, 1.2.0.0
  nanoFramework.Graphics.Core, 1.1.37.0
  nanoFramework.Hardware.Esp32, 1.6.3.0
  nanoFramework.Hardware.Esp32.Rmt, 2.0.1.0
  mscorlib, 1.14.3.0

Native Assemblies:
  mscorlib v100.5.0.18, checksum 0xCCE8376E
  nanoFramework.Runtime.Native v100.0.9.0, checksum 0x109F6F22
  nanoFramework.Hardware.Esp32 v100.0.9.0, checksum 0xC0E1540A
  nanoFramework.Hardware.Esp32.Rmt v100.0.4.0, checksum 0x608C5658
  nanoFramework.Device.Bluetooth v100.0.4.0, checksum 0xA7C14F9F
  nanoFramework.Networking.Sntp v100.0.4.4, checksum 0xE2D9BDED
  nanoFramework.ResourceManager v100.0.0.1, checksum 0xDCD7DF4D
  nanoFramework.System.Collections v100.0.1.0, checksum 0x2DC2B090
  nanoFramework.System.Text v100.0.0.1, checksum 0x8E6EB73D
  nanoFramework.Runtime.Events v100.0.8.0, checksum 0x0EAB00C9
  EventSink v1.0.0.0, checksum 0xF32F4C3E
  System.IO.FileSystem v1.0.0.0, checksum 0x3AB74021
  System.Math v100.0.5.4, checksum 0x46092CB1
  System.Net v100.1.5.0, checksum 0x5BAB8CB3
  System.Device.Adc v100.0.0.0, checksum 0xE5B80F0B
  System.Device.Gpio v100.1.0.6, checksum 0x097E7BC5
  System.Device.I2c v100.0.0.2, checksum 0xFA806D33
  System.Device.I2s v100.0.0.1, checksum 0x478490FE
  System.Device.Pwm v100.1.0.4, checksum 0xABF532C3
  System.IO.Ports v100.1.6.1, checksum 0xB798CE30
  System.Device.Spi v100.1.2.0, checksum 0x3F6E2A7E
  System.Runtime.Serialization v100.0.0.0, checksum 0x0A066871
  System.Device.Wifi v100.0.6.4, checksum 0x00A058C6
  Windows.Storage v100.0.2.0, checksum 0x954A4192

++++++++++++++++++++++++++++++++
++        Memory Map          ++
++++++++++++++++++++++++++++++++
  Type     Start       Size
++++++++++++++++++++++++++++++++
  RAM   0x3fcae6a0  0x00032800
  FLASH 0x00000000  0x00800000

+++++++++++++++++++++++++++++++++++++++++++++++++++++++++++++++
++                   Flash Sector Map                        ++
+++++++++++++++++++++++++++++++++++++++++++++++++++++++++++++++
  Region     Start      Blocks   Bytes/Block    Usage
+++++++++++++++++++++++++++++++++++++++++++++++++++++++++++++++
      0    0x00010000       1      0x1A0000     nanoCLR
      1    0x001B0000       1      0x2E0000     Deployment
      2    0x00490000       1      0x200000     Configuration

+++++++++++++++++++++++++++++++++++++++++++++++++++
++              Storage Usage Map                ++
+++++++++++++++++++++++++++++++++++++++++++++++++++
  Start        Size (kB)           Usage
+++++++++++++++++++++++++++++++++++++++++++++++++++
  0x00490000    0x200000 (2048kB)   Configuration
  0x00010000    0x1A0000 (1664kB)   nanoCLR
  0x001B0000    0x2E0000 (2944kB)   Deployment

Deployment Map
Empty

### Description

I have one ESP32-S3 N8R8 module. When I use the latest official firmware, it can't recognize the 8MB PSRAM.

It works fine when using ESP-IDF. 

I have checked the ESP-IDF sdkconfig file, there is one config item CONFIG_SPIRAM_MODE_OCT=y . I believe this is the cause.

But I also checked the ESP-IDF documentation and the ESP32S3 manual, there are two differenct line mode of the PSRAM. For the NXR2 module, the mode config should be CONFIG_SPIRAM_MODE_QUAD=y.

### How to reproduce

Flash the latest version firmware and start, you'll see the message from the serial console.

### Screenshots

load:0x3fce3808,len:0x1320
load:0x403c9700,len:0xb88
load:0x403cc700,len:0x2ec4
entry 0x403c9940
I (24) boot: .NET nanoFramework 2nd stage bootloader ESP-IDF v4.4.5
I (25) boot: build Jul 27 2023 10:49:38
I (25) boot: chip revision: v0.1
I (29) boot.esp32s3: Boot SPI Speed : 80MHz
I (33) boot.esp32s3: SPI Mode       : DIO
I (38) boot.esp32s3: SPI Flash Size : 8MB
I (43) boot: Enabling RNG early entropy source...
I (48) boot: Partition Table:
I (52) boot: ## Label            Usage          Type ST Offset   Length
I (59) boot:  0 nvs              WiFi data        01 02 00009000 00006000
I (67) boot:  1 phy_init         RF data          01 01 0000f000 00001000
I (74) boot:  2 factory          factory app      00 00 00010000 001a0000
I (82) boot:  3 deploy           Unknown data     01 84 001b0000 002e0000
I (89) boot:  4 config           Unknown data     01 82 00490000 00200000
I (96) boot: End of partition table
I (101) esp_image: segment 0: paddr=00010020 vaddr=3c0d0020 size=2b72ch (177964)                                                                                                                                                              map
I (141) esp_image: segment 1: paddr=0003b754 vaddr=3fc99090 size=03e3ch ( 15932)                                                                                                                                                              load
I (145) esp_image: segment 2: paddr=0003f598 vaddr=40374000 size=00a80h (  2688)                                                                                                                                                              load
I (147) esp_image: segment 3: paddr=00040020 vaddr=42000020 size=c59e0h (809440)                                                                                                                                                              map
I (300) esp_image: segment 4: paddr=00105a08 vaddr=40374a80 size=14610h ( 83472)                                                                                                                                                              load
I (319) esp_image: segment 5: paddr=0011a020 vaddr=600fe000 size=0002ch (    44)                                                                                                                                                              load
I (328) boot: Loaded app from partition at offset 0x10000
I (328) boot: Disabling RNG early entropy source...
E (339) psram: PSRAM ID read error: 0x00ffffff, PSRAM chip not found or not supp                                                                                                                                                             orted, or wrong PSRAM line mode
E (340) spiram: SPI RAM enabled but initialization failed. Bailing out.
I (346) cpu_start: Failed to init external RAM; continuing without it.
I (353) cpu_start: Pro cpu up.
I (357) cpu_start: Starting app cpu, entry point is 0x403790fc
I (0) cpu_start: App cpu up.
I (379) cpu_start: Pro cpu start user code
I (379) cpu_start: cpu freq: 240000000
I (379) cpu_start: Application information:
I (382) cpu_start: Project name:     nanoCLR
I (387) cpu_start: App version:      1.8.1.358
I (392) cpu_start: Compile time:     Jul 27 2023 10:49:09
I (398) cpu_start: ELF file SHA256:  11772cd4e92034a9...
I (404) cpu_start: ESP-IDF:          v4.4.5
I (409) cpu_start: Min chip rev:     v0.0
I (413) cpu_start: Max chip rev:     v0.99
I (418) cpu_start: Chip rev:         v0.1
I (423) heap_init: Initializing. RAM available for dynamic allocation:
I (430) heap_init: At 3FCA68A0 len 00042E70 (267 KiB): D/IRAM
I (437) heap_init: At 3FCE9710 len 00005724 (21 KiB): STACK/DIRAM
I (443) heap_init: At 3FCF0000 len 00008000 (32 KiB): DRAM
I (449) heap_init: At 600FE02C len 00001FD4 (7 KiB): RTCRAM
I (457) spi_flash: detected chip: generic
I (461) spi_flash: flash io: dio
I (466) sleep: Configure to isolate all GPIO pins in sleep state
I (471) sleep: Enable automatic switching of GPIO sleep configuration
I (490) cpu_start: Starting scheduler on PRO CPU.
I (0) cpu_start: Starting scheduler on APP CPU.

josesimoes, 2023-08-02:
This should be fixed now. Can you confirm?

josesimoes, 2023-08-25:
Closing this now as there is no reply from author.
PSRAM should be OK now for S3 targets.

closed 2023-08-25 by josesimoes

marcantoinebouchard, 2024-04-24:
I have the same issue with 1.9.1.47, the 8MB PSRAM is not recognized:

.NET nanoFramework Firmware Flasher v2.5.61+d0a7ea15b0
Copyright (C) 2019 .NET Foundation and nanoFramework project contributors

Reading details from chip...OK

Connected to:
ESP32-S3 (ESP32-S3 (revision v0.1))
Features WiFi, BLE
Flash size 16MB GD25Q128 from GIGADEVICE (manufacturer 0x200 device 0x16408)
PSRAM: undetermined
Crystal 40MHz
MAC 34:85:18:8D:74:88

No target name was provided! Using 'ESP32_S3' based on the device characteristics.

Extracting ESP32_S3-1.9.1.47.zip...OK

Updating to 1.9.1.47

CoryCharlton, 2024-04-24:
Confirmed PSRAM is working on a UM FeatherS3 test device.

@marcantoinebouchard that message is not indicating that there is no PSRAM (although to be fair I'm not sure what it does indicate).

If you attach a serial debugger you should see output such as this at boot:

```
Build:Mar 27 2021
rst:0x1 (POWERON),boot:0x2a (SPI_FAST_FLASH_BOOT)
SPIWP:0xee
mode:DIO, clock div:1
load:0x3fce3808,len:0x1320
load:0x403c9700,len:0xb80
load:0x403cc700,len:0x2ec4
SHA-256 comparison failed:
Calculated: 005efcf4e8a782a09231cd42eba8f31b38401696062e6282eb2e595a23419be8
Expected: 259862281f3e7d3ea65986725278c78d86c8a8fcd0ab74b4093e071a7a931ab8
Attempting to boot anyway...
entry 0x403c993c
I (43) boot: .NET nanoFramework 2nd stage bootloader ESP-IDF v4.4.7
I (43) boot: build Apr 10 2024 16:16:26
I (43) boot: chip revision: v0.1
I (47) boot.esp32s3: Boot SPI Speed : 80MHz
I (52) boot.esp32s3: SPI Mode       : DIO
I (56) boot.esp32s3: SPI Flash Size : 16MB
I (61) boot: Enabling RNG early entropy source...
I (67) boot: Partition Table:
I (70) boot: ## Label            Usage          Type ST Offset   Length
I (78) boot:  0 nvs              WiFi data        01 02 00009000 00006000
I (85) boot:  1 phy_init         RF data          01 01 0000f000 00001000
I (93) boot:  2 factory          factory app      00 00 00010000 001a0000
I (100) boot:  3 deploy           Unknown data     01 84 001b0000 002e0000
I (108) boot:  4 config           Unknown data     01 82 00490000 00300000
I (115) boot: End of partition table
I (119) esp_image: segment 0: paddr=00010020 vaddr=3c0d0020 size=2c260h (180832) map
I (160) esp_image: segment 1: paddr=0003c288 vaddr=3fc97e50 size=03d70h ( 15728) load
I (169) esp_image: segment 2: paddr=00040000 vaddr=40374000 size=00018h (    24) load
I (169) esp_image: segment 3: paddr=00040020 vaddr=42000020 size=cb5c4h (832964) map
I (324) esp_image: segment 4: paddr=0010b5ec vaddr=40374018 size=13e30h ( 81456) load
I (347) esp_image: segment 5: paddr=0011f424 vaddr=600fe000 size=0002ch (    44) load
I (355) boot: Loaded app from partition at offset 0x10000
I (355) boot: Disabling RNG early entropy source...
I (367) cpu_start: Multicore app
I (368) spiram: Found 64MBit SPI RAM device
I (368) spiram: SPI RAM mode: sram 40m
I (369) spiram: PSRAM initialized, cache is in normal (1-core) mode.
I (376) cpu_start: Pro cpu up.
I (380) cpu_start: Starting app cpu, entry point is 0x4037940c
I (0) cpu_start: App cpu up.
I (402) cpu_start: Pro cpu start user code
I (402) cpu_start: cpu freq: 240000000
I (402) cpu_start: Application information:
I (404) cpu_start: Project name:     nanoCLR
I (409) cpu_start: App version:      0.9.99.999
I (415) cpu_start: Compile time:     Apr 23 2024 05:25:37
I (421) cpu_start: ELF file SHA256:  024b332454a2f625...
I (427) cpu_start: ESP-IDF:          v4.4.7
I (432) cpu_start: Min chip rev:     v0.0
I (436) cpu_start: Max chip rev:     v0.99 
I (441) cpu_start: Chip rev:         v0.1
I (446) heap_init: Initializing. RAM available for dynamic allocation:
I (453) heap_init: At 3FCA59A0 len 00043D70 (271 KiB): D/IRAM
I (459) heap_init: At 3FCE9710 len 00005724 (21 KiB): STACK/DIRAM
I (466) heap_init: At 3FCF0000 len 00008000 (32 KiB): DRAM
I (472) heap_init: At 600FE02C len 00001FD4 (7 KiB): RTCRAM
I (479) spiram: Adding pool of 8192K of external SPI memory to heap allocator
I (487) spi_flash: detected chip: winbond
I (491) spi_flash: flash io: dio
I (507) cpu_start: Starting scheduler on PRO CPU.
I (0) cpu_start: Starting scheduler on APP CPU.
I (507) spiram: Reserving pool of 32K of internal memory for DMA/internal allocations
```

marcantoinebouchard, 2024-04-24:
I get similar output in my serial terminal, except there are no lines mentionning "spiram". I know the external ram is not detected because the call to NativeMemory.GetMemoryInfo(NativeMemory.MemoryType.All, ...) returns about 300k. I erased the flash, and the call to nanoff --platform esp32 --serialport COM3 --devicedetails still returns PSRAM: undetermined. I am using an ESP32-S3-WROOM-1 N16R8.

marcantoinebouchard, 2024-04-24:
Micropython has 2 images for esp32s3, one with PSRAM and one without. Does nanoframework have something similar?

josesimoes, 2024-04-24:
Our default ESP32S3 is configured to have SPIRAM support enabled by default. The thing is that it's enabled for QUAD SPI not OCTAL.
I'm affraid you'll have to go for a local build for the time being...

marcantoinebouchard, 2024-04-24:
I tried with an ESP32-S3-DevKitC-1 which has 2MB spiram. Even though the nanoff command mentionned PSRAM: undetermined, the NativeMemory.GetMemoryInfo returns 2MB+internal ram.

marcantoinebouchard, 2024-04-24:
#1477