nanoframework/Home #1338: UsbClient.CreateUsbStream() does not check for too long device description

sky9mike, 2023-07-30:
### Library/API/IoT binding

System.Device.UsbStream v100.0.0.6, checksum 0x21BF24CB

### Target name(s)

Giant Gecko

### Device capabilities

System Information
HAL build info: nanoCLR running @ SKY_EEVB_Debug
  Target:   SKY_EEVB_Debug
  Platform: GGECKO_S1

Firmware build Info:
  Date:        Jun 23 2023
  Type:        MinSizeRel build with Azure RTOS v6.2.0
  CLR Version: 1.8.1.152
  Compiler:    GNU ARM GCC v12.2.1

OEM Product codes (vendor, model, SKU): 0, 0, 0

Serial Numbers (module, system):
  00000000000000000000000000000000
  0000000000000000

Target capabilities:
  Has nanoBooter: YES
  nanoBooter: v1.8.1.152
  IFU capable: NO
  Has proprietary bootloader: NO

AppDomains:

Assemblies:
  Si5575CEVBFirmware, 1.0.0.0
  nanoFramework.Runtime.Events, 1.11.6.0
  System.Device.Gpio, 1.1.28.0
  mscorlib, 1.14.3.0
  Com.Skyworks.Firmware, 1.0.0.0
  Com.SkyworksInc.NanoFramework.Devices.I2c, 1.0.101.43518
  System.Math, 1.5.29.0
  nanoFramework.System.Text, 1.2.37.0
  System.Text.RegularExpressions, 1.1.37.0
  Com.SkyworksInc.NanoFramework.Devices.Spi, 1.0.127.17255
  UnitsNet.ElectricCurrent, 5.29.0.0
  System.Runtime.Serialization, 1.0.34.0
  nanoFramework.Hardware.GiantGecko, 1.1.1.61567
  System.Buffers.Binary.BinaryPrimitives, 1.2.0.0
  System.Device.UsbStream, 1.0.46.0
  nanoFramework.Runtime.Native, 1.6.6.0
  nanoFramework.System.Collections, 1.5.18.0
  UnitsNet.ElectricPotential, 5.29.0.0
  System.IO.Streams, 1.1.38.0
  System.Diagnostics.Stopwatch, 1.2.0.0
  UnitsNet.Power, 5.29.0.0
  Com.Skyworks.Firmware.EVB, 1.0.0.0
  Com.Skyworks.Firmware.CEVB, 1.0.0.0

Native Assemblies:
  mscorlib v100.5.0.18, checksum 0xCCE8376E
  nanoFramework.Runtime.Native v100.0.9.0, checksum 0x109F6F22
  nanoFramework.Device.OneWire v100.0.4.0, checksum 0xB95C43B4
  nanoFramework.ResourceManager v100.0.0.1, checksum 0xDCD7DF4D
  nanoFramework.System.Collections v100.0.1.0, checksum 0x2DC2B090
  nanoFramework.System.Text v100.0.0.1, checksum 0x8E6EB73D
  nanoFramework.GiantGecko.Adc v100.1.0.0, checksum 0x0516D85F
  nanoFramework.Hardware.GiantGecko v100.0.0.2, checksum 0xF42EF338
  nanoFramework.Runtime.Events v100.0.8.0, checksum 0x0EAB00C9
  EventSink v1.0.0.0, checksum 0xF32F4C3E
  System.Math v100.0.5.4, checksum 0x46092CB1
  System.Device.Gpio v100.1.0.6, checksum 0x097E7BC5
  System.Device.Pwm v100.1.0.4, checksum 0xABF532C3
  System.Runtime.Serialization v100.0.0.0, checksum 0x0A066871
  System.Device.UsbStream v100.0.0.6, checksum 0x21BF24CB
  Com.SkyworksInc.NanoFramework.Devices.I2c v100.0.0.2, checksum 0x913E3D1E
  Com.SkyworksInc.NanoFramework.Devices.Spi v100.0.0.7, checksum 0x060333BF

++++++++++++++++++++++++++++++++
++        Memory Map          ++
++++++++++++++++++++++++++++++++
  Type     Start       Size
++++++++++++++++++++++++++++++++
  RAM   0x20000000  0x00080000
  FLASH 0x00000000  0x00100000

+++++++++++++++++++++++++++++++++++++++++++++++++++++++++++++++
++                   Flash Sector Map                        ++
+++++++++++++++++++++++++++++++++++++++++++++++++++++++++++++++
  Region     Start      Blocks   Bytes/Block    Usage
+++++++++++++++++++++++++++++++++++++++++++++++++++++++++++++++
      0    0x00000000      12      0x001000     nanoBooter
      1    0x0000C000      55      0x001000     nanoCLR
      2    0x00043000     444      0x001000     Deployment
      3    0x001FF000       1      0x001000     Configuration

+++++++++++++++++++++++++++++++++++++++++++++++++++
++              Storage Usage Map                ++
+++++++++++++++++++++++++++++++++++++++++++++++++++
  Start        Size (kB)           Usage
+++++++++++++++++++++++++++++++++++++++++++++++++++
  0x00000000    0x00C000 (48kB)     nanoBooter
  0x001FF000    0x001000 (4kB)      Configuration
  0x0000C000    0x037000 (220kB)    nanoCLR
  0x00043000    0x1BC000 (1776kB)   Deployment

Deployment Map
Empty

### Description

I had "XXXXXXXX XXX6900x/XXXXx/XXXXx/XXXXx CEVB" but saw a hard crash. X is to redact product information.

The managed code should have thrown an ArgumentOutOfRangeException instead. The documentation of the parameter should list the limit. 
[image]

sky9mike, 2023-07-30:
Does the binding may have too small a size limit vs. WinUSB limit? See https://stackoverflow.com/a/7283272.

josesimoes, 2023-07-31:
@sky9mike yes, this is currently hard coded limit to 32 chars. You know: saving RAM space and all that.

Agreed that it should validate (and show on the docs) the lenght of that parameter.
As we are on this: what is your take on the 32 chars lenght? OK? Should be increased? Do we need to reserve those 256 chars lenght?

sky9mike, 2023-07-31:
If not too much trouble, make it a macro that can be overridden in target folder and document that. A default of 64 char sounds good.

closed 2023-08-03 by josesimoes