nanoframework/Home #1339: System.Net.Http library failing to send more than 16 HTTP requests

thisRandomm, 2023-08-07:
### Tool

Visual Studio extension

### Description

I got the library to work and send HTTP requests over WiFi to my computer, however, whenever I try to send more than 16 requests, I get the following errors:

With POST requests:
```
Exception thrown: 'System.Net.Sockets.SocketException' in System.Net.Http.dll
    ++++ Exception System.Net.Sockets.SocketException - CLR_E_FAIL (1) ++++
    ++++ Message: 
    ++++ System.Net.Sockets.NativeSocket::socket [IP: 0000] ++++
    ++++ System.Net.Sockets.Socket::.ctor [IP: 0024] ++++
    ++++ System.Net.HttpWebRequest::EstablishConnection [IP: 0141] ++++
    ++++ System.Net.HttpWebRequest::SubmitRequest [IP: 001a] ++++
    ++++ System.Net.HttpWebRequest::GetRequestStream [IP: 0008] ++++
    ++++ System.Net.Http.HttpMessageInvoker::Send [IP: 0013] ++++
    ++++ System.Net.Http.HttpClient::Send [IP: 006f] ++++
    ++++ System.Net.Http.HttpClient::Post [IP: 0011] ++++
    ++++ NFApp1.Program::Main [IP: 0037] ++++
```

With GET requests:
```
Exception thrown: 'System.Net.Http.HttpRequestException' in System.Net.Http.dll
    ++++ Exception System.Net.Sockets.SocketException - CLR_E_FAIL (1) ++++
    ++++ Message: 
    ++++ System.Net.Sockets.NativeSocket::socket [IP: 0000] ++++
    ++++ System.Net.Sockets.Socket::.ctor [IP: 0024] ++++
    ++++ System.Net.HttpWebRequest::EstablishConnection [IP: 0141] ++++
    ++++ System.Net.HttpWebRequest::SubmitRequest [IP: 001a] ++++
    ++++ System.Net.HttpWebRequest::GetResponse [IP: 000c] ++++
    ++++ System.Net.Http.HttpClientHandler::Send [IP: 00a7] ++++
    ++++ System.Net.Http.HttpMessageInvoker::Send [IP: 0013] ++++
    ++++ System.Net.Http.HttpClient::Send [IP: 006f] ++++
    ++++ System.Net.Http.HttpClient::Get [IP: 000c] ++++
    ++++ Exception System.Net.WebException - 0x00000000 (1) ++++
    ++++ Message: GetResponse() failed
    ++++ System.Net.HttpWebRequest::GetResponse [IP: 00da] ++++
    ++++ System.Net.Http.HttpClientHandler::Send [IP: 00a7] ++++
    ++++ System.Net.Http.HttpMessageInvoker::Send [IP: 0013] ++++
    ++++ System.Net.Http.HttpClient::Send [IP: 006f] ++++
    ++++ System.Net.Http.HttpClient::Get [IP: 000c] ++++
    ++++ Exception System.Net.Http.HttpRequestException - 0x00000000 (1) ++++
    ++++ Message: An error occurred while sending the request
    ++++ System.Net.Http.HttpClientHandler::Send [IP: 00bf] ++++
    ++++ System.Net.Http.HttpMessageInvoker::Send [IP: 0013] ++++
    ++++ System.Net.Http.HttpClient::Send [IP: 006f] ++++
    ++++ System.Net.Http.HttpClient::Get [IP: 000c] ++++
```

Those errors are thrown after exacly 16 successful requests everytime. I tried disposing the HttpClient by placing it in the using statement, using just the client library instead of the full API, and I've also tried to change some HttpClient settings, however nothing helped and the issue still occurs after exacly 16 requests.
My board is ESP32 DEVKITV1

I asked about it on the nanoFramework discord, and it has been confirmed that it is not an issue on my side, as it has been replicated by another member
Link to the discord message: https://discord.com/channels/478725473862549535/481782754674212867/1137833911158853743

### How to reproduce

1. Setup a loop with a http GET request, something like this:
```
    public class Program
    {
        public static void Main()
        {
            const string SSID = "XYZ";
            const string password = "XYZ";

            CancellationTokenSource cs = new(60000);
            var success = WifiNetworkHelper.ConnectDhcp(SSID, password, requiresDateTime: true, token: cs.Token);
            Debug.WriteLine(success ? "Successfully connected to wifi" : "Failed to connect to wifi");

            for (int i = 0; i < 30; i++)
            {
                try
                {
                    using (var httpClient = new HttpClient())
                    {
                        var response = httpClient.Get("http://www.example.com/");
                        Debug.WriteLine(response.Content.ReadAsString());
                        response.Dispose();
                    }
                }
                catch (Exception e)
                {
                    Debug.WriteLine($"message: {e.Message}");
                }

                Thread.Sleep(1000);
            }

            Thread.Sleep(Timeout.Infinite);
        }
    }
```
2. Wait until it sends 16 packets, and the errors should start showing up

### Expected behaviour

It should continue sending new requests as usual

### Aditional context

Visual studio version: 17.6.3
nanoFramework.CoreLibrary version: 1.14.2
nanoFramework.System.Net version: 1.10.52
nanoFramework.System.Net.Http version: 1.5.97
.NET nanoFramework Firmware Flasher v2.5.34+c89f1d1d65

alberk8, 2023-08-07:
From a cursory check on the HttpClient.  It looks like the Socket is not disposed off.

closed 2023-08-08 by Ellerbach

f3t3l, 2024-02-16:
Hello,

This issue is closed, but still appears to be haporninh on the latest version of the system.net.http libary (1.5.125).