nanoframework/Home #1340: Network socket not working correctly when multiple network interfaces

AdrianSoundy, 2023-08-09:
### Target name(s)

ESP32

### Firmware version

All

### Was working before? On which version?

no

### Description

The socket code always checks the status of the first network interface even if call is for another interface.

lwip_sockets.c

* LWIP_SOCKETS_Driver::Select()

If  exception fds passed then looks at g_LWIP_SOCKETS_Driver.m_interfaces[0] to see if network interface is up.
This check may not need to be present if lwip does its own check. 

* GetAddrInfo()
In the case where name & service are null returns IP address local interface 0 instead of active interface
Probably this function should be separated and interface number passed in.

### How to reproduce

1. Using firmware with  wireless client and Soft AP interfaces
2. Make a connection on SoftAP and try to communicate
3. Select() calls will return network down error

### Expected behaviour

Uses correct Interface

AdrianSoundy, 2023-08-09:
This is mainly an issue on ESP32 because of the multiple network interfaces.

Ellerbach, 2023-08-09:
Any idea where to adjust the code? So we know where to start looking at?

AdrianSoundy, 2023-08-09:
Yes i know where to adjust code in the select() function of lwip_sockets.cpp.
I first need to do some testing to see if we actually need the existing code in the function and not that the latest lwip versions already handles it. It may just be historic.

If we do still need code then we need to look at all handles passed in select arrays and check if assoicatted interfaces to handles are active/up. 

Hopefully do some testing today.
@Ellerbach

AdrianSoundy, 2023-08-10:
For a socket there is not really an association to any interface. If you think about UDP send it would be routed to the correct interface based on target IP address.

So the conclusion of looking into this and some testing. We only need to make sure at least 1 interface is active.  So we can scan all configured interfaces and return error if there are no interfaces UP.

For GetAddrInfo() we need to change it to return all local addresses of interfaces that are UP, not just first interface.

alberk8, 2023-08-10:
One question, In the SoftAP where a client is connected to the AP and open a Socket connection.   So when the client disconnect via changing to another AP, does the open socket connection get closed or terminated when the SoftAP connection disassociated?

AdrianSoundy, 2023-08-10:
> One question, In the SoftAP where a client is connected to the AP and open a Socket connection. So when the client disconnect via changing to another AP, does the open socket connection get closed or terminated when the SoftAP connection disassociated?

It should do but i don't know if it does. I can test that situation. 
Its not to do with current change as SoftAP interface is still shows as UP.

AdrianSoundy, 2023-08-11:
Testing GetAddrInfo()

Tests with Wifi and SoftAP interfaces active
Test with just SoftAP

Other fixes
- Noticed that it didn't return the HostName(canon name) in IPHostEntry but supported in all other stack layes, fixed
- If NULL passed in for host name caused ESP32 to crash - fixed
 
Test code
```
IPHostEntry he = Dns.GetHostEntry("");
Console.WriteLine($"Host name {he.HostName}");
foreach(IPAddress adr in he.AddressList)
{
    Console.WriteLine( adr.ToString() );
}

he = Dns.GetHostEntry("google.com");
Console.WriteLine($"Host name : {he.HostName}");
foreach (IPAddress adr in he.AddressList)
{
    Console.WriteLine(adr.ToString());
}
```    

Now returns an array of IPAddress for each configured  local Interface when hostname = ""
Correctly returns canon name of host name when doing a DNS lookup.

AdrianSoundy, 2023-08-12:
The Select code is called by the Poll() logic in the System.Net for each handle.  Code change to just test for at least 1 interface up instead of just first interface (NETIF) otherwise Network down error reported.  

ESP32 platforms:
Also noticed that if a link goes up/down the Poll wasn't signaled via the via the EVENT_SOCKET & EVENT_NETWORK so would just wait.
Other platforms use the LWIP link and callback but using these on ESP32 causes various exceptions. I think it's because ESP_IDF is already using them for generating its own esp32 network events. Add call to set events when network link goes up/down or address changes.

Ellerbach, 2023-08-14:
Seems like great progress!

AdrianSoundy, 2023-08-15:
Another change was done to set the Socket & Network events when a connect/disconnect events happens.  On other platforms we are able to use the LWIP_NETIF_LINK_CALLBACK & LWIP_NETIF_STATUS_CALLBACK but for ESP32 this seems to cause crashes, memory corruption issues. If maybe because ESP_IDF is already link to this for event handler.

AdrianSoundy, 2023-08-15:
Fix:-
While testing with WifiAP sample on a firmware with Ethernet found Ethernet would crash and stop.
Found that sample uses GPIO 5 for an input button which happens to be used by Ethernet PHY.
Changed Ethernet driver to reserve all GPIO pins it so C# will report error when a GPIO is used.

closed 2023-08-16 by AdrianSoundy