nanoframework/Home #1341: Add Support for Arduino Nano ESP32

AslamNazeerShaikh, 2023-08-10:
### Description

Add Support for Arduino Nano ESP32

### How to solve the problem

This board brings all the power of the well-known ESP32-S3 with the full support of the Arduino ecosystem.

### Describe alternatives you've considered

Add Support for Arduino Nano ESP32

### Aditional context

Add Support for Arduino Nano ESP32

alberk8, 2023-08-11:
@AslamNazeerShaikh , From the specs it looks like it can be supported by current ESP32_S3 firmware.

AslamNazeerShaikh, 2023-08-14:
Arduino Nano ESP32 Documentation:
1. https://docs.arduino.cc/resources/schematics/ABX00083-schematics.pdf
2. https://docs.arduino.cc/resources/datasheets/ABX00083-datasheet.pdf

alberk8, 2023-08-14:
A lot of the specification from the MCU manufacturer (espressif) are the highest possible value but you need to read the fine print.  For example the Analog I/O if you use all the 20 then a lot of other peripheral will not work, same for PWM, etc.  because alot of I/O has multiple Modes but only one at any one time.  

BTW, I don't have the device so I cannot test.

AslamNazeerShaikh, 2023-08-24:
Hi NanoFramework Team,

Please lets us know that can we get official support for -> "Arduino Nano ESP32"

josesimoes, 2023-08-24:
@AslamNazeerShaikh as mentioned by Albert, you should be able to use the ESP32_S3 firmware with this board.
Any changes required in the GPIO configuration can be handled by software by reasigning the GPIOs to new functions.

josesimoes, 2023-09-01:
@AslamNazeerShaikh have you tested the ESP32_S3 firmware with this board?

AslamNazeerShaikh, 2023-10-07:
Hi Team, Hi @josesimoes . First of all, sorry for the late response. 

Sure I need some time to test this firmware on this board. Currently I do not have this board to test it.
I live in India and "Adriano Uno ESP32 S3" board having huge shortage in my country/region right now.
I would to like to inform our nano community that it will take me next 1 to 2 months to get a copy of this board.

Sure I will let the team know about the final results soon.

AslamNazeerShaikh, 2024-01-27:
Closing the issue as of now, as I don't have access to the above board. 
Also, I request anyone can re-open current or new issue for the above same topic if they want to contribute on this. 
Thanks!

closed 2024-01-27 by AslamNazeerShaikh