nanoframework/Home #1343: Add support for ESP32 with 32MB flash

mikeobrien, 2023-08-12:
### Description

I just picked up an [ESP32­-S3­-DevKitC-1 with 32MB flash from adafruit](https://www.adafruit.com/product/5364) and when I go to flash the ESP32-S3 target I get the following error:

```
There is no firmware available for ESP32 with 32MB flash size!
Only the following flash sizes are supported: 2MB, 4MB, 8MB, 16MB
Error E4001: Unsupported flash size for ESP32 target.
```

Also, if you you don't specify `-v d` you only get the following error which can be confusing:

```
E4001
```

### How to solve the problem

- Add support to the ESP32_S3 target for 32MB flash.
- Display the full error message in normal verbosity.

### Describe alternatives you've considered

- Buy a 16MB version.
- Specify detailed verbosity.

josesimoes, 2023-09-01:
Yes, support for this memory size can be added. FYI you won't be able to use it completely as the larger sizes require bank switching to address the upper addresses which is something we don't have any plans to support.

AdrianSoundy, 2023-09-11:
I'll include this change with the idf 5.1.1 update

pracsol, 2023-10-06:
At risk of asking a dumb q, is this support dependent upon issue 1345? New to area, but curious if this issue marked as trivial is gated by the issue marked as non-trivial.. deciding whether to buy a 16MB board or wait. :D

josesimoes, 2023-10-07:
@pracsol answering the fundamental question there: go for the 16M. Over that size it's useless as it requires bank switching to access it. Not practical nor implemented in nanoFramework. 😉

networkfusion, 2024-05-09:
Please confirm fixed. @AdrianSoundy.

AdrianSoundy, 2024-06-08:
Currently not fixed. The S3 target currently supports up to 16Mb flash.

I think the bank switching was to do with PSRAM not flash.  @josesimoes 
I have noticed with S3 targets you get 8Mb psram when available were this was limited to 4MB in esp32

DavidH102, 2024-12-04:
This issue seems to not be resolved. test on [ESP32-S3-DevKitC-1-N32R8V Development Board ](https://a.co/d/a680xBr) I received the same Error E4001: Unsupported flash size for ESP32 target.
[image]
as a work around I am able to set it to 16mb thus:
[image]
to note I found out since that if you use --target then --platform is not needed
if there anything I can do to help diagnose this issue?

AdrianSoundy, 2024-12-05:
The S3 firmware is being built with the 32mb partition tables but it looks like "nanoff" doesn't support it.

Esp32Firmware.cs
```
/// <summary>
/// ESP32 nanoCLR is available for 2MB, 4MB, 8MB and 16MB flash sizes
/// </summary>
private List<int> SupportedFlashSizes => [0x200000, 0x400000, 0x800000, 0x1000000];
```

32MB and 64mb should be added as they are now available.

closed 2024-12-06 by AdrianSoundy