nanoframework/Home #1348: Add support for CAN bus for ESP32

robota10, 2023-08-20:
### Description

When using the CAN bus under EPS32, the target is not supported error.

System Information
HAL build info: nanoCLR running @ ESP32 built with ESP-IDF ac5d805
  Target:   ESP32_BLE_REV3
  Platform: ESP32

Firmware build Info:
  Date:        Aug  8 2023
  Type:        MinSizeRel build, chip rev. 3, support for PSRAM, support for BLE
  CLR Version: 1.8.1.423
  Compiler:    GNU ARM GCC v8.4.0

OEM Product codes (vendor, model, SKU): 0, 0, 0

Serial Numbers (module, system):
  00000000000000000000000000000000
  0000000000000000

Target capabilities:
  Has nanoBooter: NO
  IFU capable: NO
  Has proprietary bootloader: YES

AppDomains:

Assemblies:

Native Assemblies:
  mscorlib v100.5.0.18, checksum 0xCCE8376E
  nanoFramework.Runtime.Native v100.0.9.0, checksum 0x109F6F22
  nanoFramework.Hardware.Esp32 v100.0.9.0, checksum 0xC0E1540A
  nanoFramework.Hardware.Esp32.Rmt v100.0.4.0, checksum 0x608C5658
  nanoFramework.Device.Bluetooth v100.0.5.0, checksum 0xD163A9D5
  nanoFramework.Device.OneWire v100.0.4.0, checksum 0xB95C43B4
  nanoFramework.Networking.Sntp v100.0.4.4, checksum 0xE2D9BDED
  nanoFramework.ResourceManager v100.0.0.1, checksum 0xDCD7DF4D
  nanoFramework.System.Collections v100.0.1.0, checksum 0x2DC2B090
  nanoFramework.System.Text v100.0.0.1, checksum 0x8E6EB73D
  nanoFramework.Runtime.Events v100.0.8.0, checksum 0x0EAB00C9
  EventSink v1.0.0.0, checksum 0xF32F4C3E
  System.IO.FileSystem v1.0.0.0, checksum 0x3AB74021
  System.Math v100.0.5.4, checksum 0x46092CB1
  System.Net v100.1.5.0, checksum 0x5BAB8CB3
  System.Device.Adc v100.0.0.0, checksum 0xE5B80F0B
  System.Device.Dac v100.0.0.6, checksum 0x02B3E860
  System.Device.Gpio v100.1.0.6, checksum 0x097E7BC5
  System.Device.I2c v100.0.0.2, checksum 0xFA806D33
  System.Device.I2s v100.0.0.1, checksum 0x478490FE
  System.Device.Pwm v100.1.0.4, checksum 0xABF532C3
  System.IO.Ports v100.1.6.1, checksum 0xB798CE30
  System.Device.Spi v100.1.2.0, checksum 0x3F6E2A7E
  System.Runtime.Serialization v100.0.0.0, checksum 0x0A066871
  System.Device.Wifi v100.0.6.4, checksum 0x00A058C6
  Windows.Storage v100.0.2.0, checksum 0x954A4192

2>Getting things ready to deploy assemblies to .NET nanoFramework device: ESP32_BLE_REV3 @ COM13.
2>Deploy failed.
2>
2>***************************************************************************
2>
2>
2>The connected target does not have support for the following assembly(ies):
2>
2>    'nanoFramework.Device.Can'
2>
2>Please check: 
2>  1) if the target is running the most updated image.
2>  2) if the target image was built to include support for all referenced assemblies.
2>
2>Our Visual Studio FAQ has a troubleshooting guide: https://docs.nanoframework.net/content/faq/working-with-vs-extension.html
2>
2>***************************************************************************
2>

### How to solve the problem

Add native implementation of Device.Can.

### Aditional context

[image]
[image]

Ellerbach, 2023-08-21:
> Please add a native implementation of Device.Can

.NET nanoFramework is first a community driven project. Meaning, that it's a community effort. If you really want a feature and don't know how to implement it, you can ask a professional to do it by funding this work. See here: https://docs.nanoframework.net/content/contributing/index.html and also the partners which could develop with confident what you need: https://www.nanoframework.net/partner-program/

robota10, 2023-08-29:
Since it was the first time to participate in an open source project and the first time to donate, I was not familiar with it. After some twists and turns, I managed to donate $50 and hope to help solve this problem.

pocki, 2024-02-03:
I would also like to see CAN Support on ESP32.
But my C/C++ knowledge is to low that I can help

robota10, 2024-03-27:
Me too. That's why I'm here for help.