nanoframework/Home #1350: SiLabs Gecko RTM CLR Doesn't Run

sky9mike, 2023-08-25:
### Target name(s)

SL_STK3701A

### Device capabilities

Target Information
HAL build info: nanoBooter running @ SKY_EEVB
  Target:     SKY_EEVB
  Platform:   GGECKO_S1
  nanoBooter: v1.8.1.203
  nanoCLR:    v1.8.1.203
  Type:       MinSizeRel build with Azure RTOS v6.2.0

### Description

While I have not actually tested w/ SL_STK3701A, our custom CLR when compiled in RTM mode will not boot and device stays in bootloader mode.

closed 2023-08-31 by josesimoes

sky9mike, 2023-08-31:
> CAUTION: The email below is from an external source. Use caution before opening attachments or clicking links.
>
> @sky9mike<https://urldefense.proofpoint.com/v2/url?u=https-3A__github.com_sky9mike&d=DwMCaQ&c=VYRDWu-sKuQrybEAJ2u-dYX_FK6X1lTrDf-PKXUa2P4&r=rU77juL7WDDQNpihqtufN1AjI4udQsWcoK9X4kgzaGs&m=Zv6VGdh-tzfuIOf6iY8uYmBiGCGkbz7NPVlj5ex0FU63TgDlj0sbTYD-7I99dBVR&s=8QaV5QYZQe3XqvrpT4v02uPhnGGhGMLXEUyezHL8MVM&e=> please note the fix in f6c1654 - Fix CLR startup params<https://urldefense.proofpoint.com/v2/url?u=https-3A__github.com_Skyworks-2DTiming-2DSoftware_nf-2Dinterpreter_commit_f6c16545d63fa366dddfcd567ab05ca7d2f8beb6&d=DwMCaQ&c=VYRDWu-sKuQrybEAJ2u-dYX_FK6X1lTrDf-PKXUa2P4&r=rU77juL7WDDQNpihqtufN1AjI4udQsWcoK9X4kgzaGs&m=Zv6VGdh-tzfuIOf6iY8uYmBiGCGkbz7NPVlj5ex0FU63TgDlj0sbTYD-7I99dBVR&s=E9zD49ex9m-uHLfelMpk2PR-IWoeUG3jrwyIJVjT3AE&e=>.
>
> The CLR startup parameters were configured to reboot to nanoBooter on application exit. This is not happening now.
> Also worth mentioning that the EnterDebuggerLoopAfterExit is now true or false depending on RTM build.
> In case of RTM this means that the board will keep rebooting until there is an application deployed.
>
> To full test these changes both referenced PRs need to be merged into Skyworks repo before you can fully test this.

Jose, can you create a document describing in detail the differences for lay-developer between RTM and Debug release CLR build? EnterDebuggerLoopAfterExit… huh for example.

Will try/catch be faster?

Thanks,
-Mike

josesimoes, 2023-09-01:
@sky9mike there is already an explanation about this RTM build option [here](https://docs.nanoframework.net/content/building/cmake-presets.html?q=NF_BUILD_RTM).
Does that cover what you were asking?

sky9mike, 2023-09-01:
That seems sufficient, thanks!

sky9mike, 2023-09-05:
I believe that has been merged into our branch @josesimoes. I see it in commits in our branch.

But when I flash, the ready light cycles very fast. Issue on both SKY_EEVB_Release and Si5575_CEVB_Release.

Did you test on our EEVB? Here is a video:

https://github.com/nanoframework/Home/assets/103005409/e96ba96f-958a-4f0b-a0d2-16ad7518313b