nanoframework/Home #1353: [DependencyInjection] Provide "Scoped" lifetime support.

jazzman55, 2023-08-29:
### Description

In some scenarios services created by DI are needed only for a limited time and scope. After they are finished, memory they consumed could be reclaimed by GC. For example:
Device runs a scheduled task e.g. every few hours. The task requires several services to perform HTTP call and execute some logic. There is also another task, triggered by pressing a button, which requires other services.
All services could be registered in DI with _Scoped_ lifetime. Then each executing task would create a scope for resolving its services. After the task finished, the scope would be disposed and all memory could be reclaimed by GC.
This is similar to ASP.NET MVC, which creates a DI scope for every request.

### How to solve the problem

Provide Scope lifetime functionality, similar to the DI in ASP.NET MVC.

burtonrodman, 2023-09-18:
@jazzman55 
I am new to nanoFramework, but very comfortable with .Net DI scenarios.  I am looking for a way to contribute something useful.  While I haven't gotten deep in my project architecture yet, this sounds like something I might want.

Could you provide a small example of how you're using DI now?

burtonrodman, 2023-09-18:
sorry, I missed that you had submitted a PR.

josesimoes, 2023-09-18:
Hey @burtonrodman welcome to nanoFramework!
Glad to see that you're already looking for a way to contribute back to the project. That's the way to go! 👍🏻 😃 

Yes, there's an open PR to tackle this. feel free to review and chime in.

Our DI library has been beautifully crafted by @bytewizer. 
If you find other improvement areas that make sense in embedded systems use cases, by all means, start a discussion around it.
We 💜 PRs!

closed 2023-10-04 by josesimoes