nanoframework/Home #1358: nanoff should allow flashing application when device is in bootloader mode

sky9mike, 2023-09-07:
### Description

Currently with Debug CLR, if you have borked your application and need to recover via USB (not J-Link), you will need to perform this multi-step process:

1. Reset MCU, asserting Stay in BL pin
2. Erase deployment area via VS.NET extension or nanoff 
3. Reset MCU, CLR now runs w/out application
4. Now I can flash via USB

But I should just be able to flash the application via USB when in BL mode.

### How to solve the problem

See above.

sky9mike, 2023-09-07:
This also applies to VS.NET extension use case.

josesimoes, 2023-09-29:
Just tested this and it's working. 
With Skyworks EVB it will fully functional once we get past #1359

sky9mike, 2023-09-29:
Woo woo!

josesimoes, 2024-03-25:
Closing as this is implemented since 2023-SEP-29.

closed 2024-03-25 by josesimoes