nanoframework/Home #1362: VSIX extension fails to overwrite a malformed network configuration

thwr, 2023-09-11:
### Tool

Visual Studio extension

### Description

The VSIX extension seems to run into a situation when it encounters an invalid network configuration read from the attached board. In this case, it's not possible anymore to set an MAC address for an existing interface, it's also not possible to set a MAC by code.

The code in question is not inside the VSIX extension, [but in the nf-debugger repo](https://github.com/nanoframework/nf-debugger/blob/main/nanoFramework.Tools.DebugLibrary.Shared/WireProtocol/Engine.cs#L4435), which is consumed as nuget by the VSIX extension

### How to reproduce

Not sure, this seem to have happened when I tried to upload a PEM file on two identical boards using the network configuration dialog.

### Expected behaviour

It would be way better to have the VSIX extension to write back some defaults instead of just failing.

### Screenshots

n/a

### Aditional context

Latest VSIX 2022.3.0.60
Fully patched VS2022

thwr, 2023-09-11:
See also this thread: https://discord.com/channels/478725473862549535/1150804608915095572

josesimoes, 2023-09-12:
@thwr OK to assign this to you?

thwr, 2023-09-12:
> @thwr OK to assign this to you?

I'll try my best

josesimoes, 2023-10-12:
Reproduced with NESHTEC_NESHNODE_V1 target.
The root cause is that the current build is not creating the "empty" configuration that serves as a placeholder for VS to grab and overwrite whatever it has to. 

I'm inclined to fix this issue and leave VS as it is. We could try to improve it, or add some validation on top of what's already there, but honestly I don't think it's worth the effort...

closed 2023-10-13 by josesimoes