nanoframework/Home #1364: HttpClient error

callumeveratt, 2023-09-14:
### Library/API/IoT binding

nanoframework.System.Net.Http

### Visual Studio version

N/A

### .NET nanoFramework extension version

N/A

### Target name(s)

ESP32 (ESP32-D0WDQ6 (revision v1.0))

### Firmware version

1.8.1.548

### Device capabilities

Connected to: 
ESP32 (ESP32-D0WDQ6 (revision v1.0))
Features WiFi, BT, Dual Core, 240MHz, VRef calibration in efuse, Coding Scheme None
Flash size 4MB unknown from  (manufacturer 0x104 device 0x16406)
PSRAM: not available
Crystal 40MHz

### Description

We have a while loop which should run a get request using a HttpClient every 15 minutes. Each time the first request works, then each subsequent request will fail with the below error. I thought this may be related to #1339 but I am running the latest version of the nuget package.

Booting up 
API Responded: true
-- NEW REQUEST --
    ++++ Exception System.Net.Sockets.SocketException - CLR_E_FAIL (1) ++++
    ++++ Message: 
    ++++ System.Net.Security.SslNative::SecureClientInit [IP: 0000] ++++
    ++++ System.Net.Security.SslStream::Authenticate [IP: 0045] ++++
    ++++ System.Net.Security.SslStream::AuthenticateAsClient [IP: 000a] ++++
    ++++ System.Net.HttpWebRequest::EstablishConnection [IP: 027c] ++++
    ++++ System.Net.HttpWebRequest::SubmitRequest [IP: 001a] ++++
    ++++ System.Net.HttpWebRequest::GetResponse [IP: 000c] ++++
    ++++ System.Net.Http.HttpClientHandler::Send [IP: 00a7] ++++
    ++++ System.Net.Http.HttpMessageInvoker::Send [IP: 0013] ++++
    ++++ System.Net.Http.HttpClient::Send [IP: 006f] ++++
    ++++ System.Net.Http.HttpClient::Get [IP: 000c] ++++
    ++++ WheelieWatch.Device.Services.HttpRequest::CheckRequest [IP: 002c] ++++
    ++++ WheelieWatch.Device.Program::Main [IP: 00ba] ++++
    ++++ Exception System.Net.WebException - 0x00000000 (1) ++++
    ++++ Message: GetResponse() failed
    ++++ System.Net.HttpWebRequest::GetResponse [IP: 00da] ++++
    ++++ System.Net.Http.HttpClientHandler::Send [IP: 00a7] ++++
    ++++ System.Net.Http.HttpMessageInvoker::Send [IP: 0013] ++++
    ++++ System.Net.Http.HttpClient::Send [IP: 006f] ++++
    ++++ System.Net.Http.HttpClient::Get [IP: 000c] ++++
    ++++ WheelieWatch.Device.Services.HttpRequest::CheckRequest [IP: 002c] ++++
    ++++ WheelieWatch.Device.Program::Main [IP: 00ba] ++++
    ++++ Exception System.Net.Http.HttpRequestException - 0x00000000 (1) ++++
    ++++ Message: An error occurred while sending the request
    ++++ System.Net.Http.HttpClientHandler::Send [IP: 00bf] ++++
    ++++ System.Net.Http.HttpMessageInvoker::Send [IP: 0013] ++++
    ++++ System.Net.Http.HttpClient::Send [IP: 006f] ++++
    ++++ System.Net.Http.HttpClient::Get [IP: 000c] ++++
    ++++ WheelieWatch.Device.Services.HttpRequest::CheckRequest [IP: 002c] ++++
    ++++ WheelieWatch.Device.Program::Main [IP: 00ba] ++++
Exception thrown: 'System.Net.Http.HttpRequestException' in System.Net.Http.dll
An unhandled exception of type 'System.Net.Http.HttpRequestException' occurred in System.Net.Http.dll
Additional information: An error occurred while sending the request

### Sample project or code

// Program.cs
<code>while (true) 
					{
						HttpRequest.CheckRequest();
						Thread.Sleep(900000);
					}</code>

// HttpRequest.CheckRequest
<code>HttpClient HttpClient = new HttpClient();
			HttpClient.HttpsAuthentCert = new X509Certificate(@"-----CERTIFICATE REMOVED FOR BREVITY-----");
		var url = "https://localhost:8080/CheckDay";
			var response = HttpClient.Get(url);
			var responseBody = response.Content.ReadAsString();
			Debug.WriteLine("API Responded: " + responseBody);</code>

josesimoes, 2023-09-14:
What device are you connecting to?

josesimoes, 2023-09-14:
Also, have you tried to add the option to close the socket after the request?

torbacz, 2023-09-14:
Also, have you tried to keep only one HttpClient instead of creating it each time?

callumeveratt, 2023-09-14:
Thanks both - tried the HttpClient.Dispose() on the original code - same issue still.

I also started with a single usage httpClient but I hit a different error with that - I may be missing something though:
<code>
		static HttpClient _httpClient = new HttpClient()
		{
			HttpsAuthentCert = new X509Certificate(@"CERT FOR BREVITY")
	};
</code>

    ++++ Exception System.NullReferenceException - CLR_E_NULL_REFERENCE (2) ++++
    ++++ Message: 
    ++++ System.Net.WebHeaderCollection::GetValues [IP: 0007] ++++
    ++++ System.Net.Http.Headers.HttpRequestHeaders::set_ConnectionClose [IP: 000a] ++++
    ++++ WheelieWatch.Device.Services.HttpRequest::.cctor [IP: 0003] ++++
Exception thrown: 'System.NullReferenceException' in System.Net.Http.dll
An unhandled exception of type 'System.NullReferenceException' occurred in System.Net.Http.dll

torbacz, 2023-09-14:
Just create it before while loop and pass to CheckRequest method.

callumeveratt, 2023-09-14:
I am creating it outside the while loop - it's created within the HttpRequest class at the moment. I have tried it within the Program class too - and same error.

torbacz, 2023-09-14:
According to your code snipper, you are creating it each time method is executed.
[image]

callumeveratt, 2023-09-14:
So I had that in originally yes, as creating a single instance actually brings back a different error. 

> Thanks both - tried the HttpClient.Dispose() on the original code - same issue still.
> 
> I also started with a single usage httpClient but I hit a different error with that - I may be missing something though: ` static HttpClient _httpClient = new HttpClient() { HttpsAuthentCert = new X509Certificate(@"CERT FOR BREVITY") }; `
> 
> ```
> ++++ Exception System.NullReferenceException - CLR_E_NULL_REFERENCE (2) ++++
> ++++ Message: 
> ++++ System.Net.WebHeaderCollection::GetValues [IP: 0007] ++++
> ++++ System.Net.Http.Headers.HttpRequestHeaders::set_ConnectionClose [IP: 000a] ++++
> ++++ WheelieWatch.Device.Services.HttpRequest::.cctor [IP: 0003] ++++
> ```
> 
> Exception thrown: 'System.NullReferenceException' in System.Net.Http.dll An unhandled exception of type 'System.NullReferenceException' occurred in System.Net.Http.dll

callumeveratt, 2023-09-14:
Okay - so I've changed it to a single httpClient, as suggested. This is my HttpRequestService.cs file.

```
namespace CE.Device.Services
{
	public static class HttpRequest
	{
		static HttpClient _httpClient = new HttpClient()
		{
			HttpsAuthentCert = new X509Certificate(@"--------")
		};

		public static void CheckRequest()
		{
			var url = "https://localhost/CheckDate";
			var response = _httpClient.Get(url);
			response.EnsureSuccessStatusCode();
			var responseBody = response.Content.ReadAsString();
			Debug.WriteLine("API Responded: " + responseBody);
		}
	}
}
```

Then called via Program.cs, unchanged:
```
while (true)
{
	HttpRequest.CheckRequest();
	Thread.Sleep(60000);
}
```

The error now:
++++ Exception System.NullReferenceException - CLR_E_NULL_REFERENCE (2) ++++
++++ Message: 
++++ System.Net.WebHeaderCollection::GetValues [IP: 0007] ++++
++++ System.Net.Http.Headers.HttpRequestHeaders::set_ConnectionClose [IP: 000a] ++++
++++ WheelieWatch.Device.Services.HttpRequest::.cctor [IP: 0003] ++++