nanoframework/Home #1370: How can I read/write settings in runtime with settings in memory

SlimRG, 2023-09-19:
### Description

I want to have possibility to read/write settings (different types of vars) in memory.
How can I do it (don't find it in samples)?
I have ESP32-WROOM

### How to solve the problem

Maybe, create lz4 compressed binary file with strict and lazy (async) saving in non-volatile memory.

SlimRG, 2023-09-19:
I'm personal user that wants to create Hi-Tech own home :)

torbacz, 2023-09-19:
The fastest way - serialize it to JSON and store as text file on internal "I" drive. I've got working solution with such approach. Configuration can updated in runtime. 
Another topic is how you want to update settings. Few solutions, MQTT/WebServer/AzureIoTHub - up to you. 
All mentioned solutions can be found in Samples repository, just put it together. 

Happy coding :)

Ellerbach, 2023-10-04:
@SlimRG did you find one of the way @torbacz mentioned working for you? The samples repo has examples for all this!

networkfusion, 2023-10-08:
Closing as answered. Feel free to reopen if required.

closed 2023-10-08 by networkfusion