nanoframework/Home #1371: The esp32s3 device cannot be found

williamlzw, 2023-09-20:
### Tool

Visual Studio extension

### Description

esp-idf-tools-setup-offline-4.4.5.exe
esp32-s3-DevKitC-1 v1.6
visual studio2019 v16.11.26

Device Explorer cannot found device
00:33:30.617 [NanoDevices: new device arrival COM3]
00:33:30.617 [NanoDevices: candidate nano device COM3]
00:33:30.685 [NanoDevices: checking device  COM3 @ 921600]
00:33:31.707 [NanoDevices: checking device  COM3 @ 460800]
00:33:32.736 [NanoDevices: checking device  COM3 @ 115200]
00:33:35.704 [NanoDevices: checking device  COM3 *** 2nd attempt ***]
00:33:35.770 [NanoDevices: checking device  COM3 @ 921600]
00:33:37.795 [NanoDevices: checking device  COM3 @ 460800]
00:33:39.825 [NanoDevices: checking device  COM3 @ 115200]
00:33:42.412 [NanoDevices: quitting device COM3]
00:33:42.413 [NanoDevices: Serial device enumeration completed. Found 0 devices]

alberk8, 2023-09-21:
If your S3 has 2 USB port then you need to use the other USB port after flashing is done for the VS to able to detect.

williamlzw, 2023-09-21:
I re-plugged the usb cable, and the software couldn't detect the device.

alberk8, 2023-09-22:
For S3 DevKitC , the right USB is for flashing firmware and the left USB is for app loading via Visual Studio 2022

williamlzw, 2023-09-22:
esp32s2 cannot be detected in vs2019

alberk8, 2023-09-22:
It is working on my VS2019 and VS2022.  Make sure you have the latest VS extension and the serial port is also latest.  You get better support in our discord [server](https://discord.gg/net-nanoframework-478725473862549535).

williamlzw, 2023-09-22:
esp32s3 can only be found by visual studio2022.

closed 2023-09-22 by williamlzw