nanoframework/Home #1372: Bluetooth not working on ESP32-S3-DevkitC1

Alex-111, 2023-09-22:
### Target name(s)

ESP32-S3-DevkitC1

### Firmware version

ESP32_S3_BLE

### Description

I tried the sample code. BluetoothLESample2. (Also tried BluetoothLESample1) It seems that it never goes beyond that line:

line 79: server.Start()

Bug was confirmed by:
AlbertK — 19.09.2023 03:16
Confirm the Bluetooth is not working in S3.  It is rebooting/system panic.  Can you raise an issue in github so that it can be tracked?.  Thanks. 

### How to reproduce

start BluetoothLESample2

h2soft-de, 2023-10-15:
Same for ESP32 Dev KitC V4. Tested with Scanner Example. No more response from module after calling start()

Alex-111, 2023-12-02:
Just want to ask if there is another ESP32 board known to work with BLE?

W3THS, 2024-01-17:
I'd like to add a +1 to this issue as I'm also experiencing it.

agrosiorl, 2024-01-17:
Not working on:
 - Heltec LoRa 32 V3 (ESP32-S3)
 - ESP32-S3-WROOM-1

philethius, 2024-01-20:
Not working on:

- YD-ESP32-23 (ESP32-S3-WROOM-1 N16R8)

Alex-111, 2024-04-02:
The issue is still there. Just tested with the latest version. I.e.the BLE Image for S3 is useless!

networkfusion, 2024-05-09:
Please let us know if this is still an issue.

Alex-111, 2024-05-13:
What commit should fix this? I cannot see any fix or commit related to this.

torbacz, 2024-05-13:
Migration to ESP-IDF 5.x

Alex-111, 2024-05-17:
@W3THS @agrosiorl @philethius @h2soft-de 
OK - I see. Have no access to hardware the next weeks. Maybe anybody else can do a quick check for now?

Existrow, 2024-05-19:
The problem is relevant. Also reproducible

Existrow, 2024-05-19:
The wifi connection is also not working

philethius, 2024-05-19:
I had a version that worked fine after the ESP-IDF 5.x update.  However, if there's been a change since, I can no longer update any of my devices.

torbacz, 2024-05-28:
ESP32_S3_ALL-1.9.1.151
Works both - BT and WiFi

githubmafio, 2024-05-30:
Yes, it works now

Ellerbach, 2024-05-31:
So closing this issue as it seems to work. Thanks for the confirmation.

closed 2024-05-31 by Ellerbach