nanoframework/Home #1374: NetworkHelper.SetupNetwork cannot run more than once

networkfusion, 2023-09-29:
### Library/API/IoT binding

System.Net

### Target name(s)

STM32 targets

### Firmware version

latest

### Description

The following will cause an exception

```
var success = false;

while (!success)
{
    CancellationTokenSource cs = new CancellationTokenSource(1000);
    success = NetworkHelper.SetupAndConnectNetwork(requiresDateTime: true, token: cs.Token);
    Thread.Sleep(1000);
}
```

If the network takes more than one loop.

### Expected behaviour

You should be able to call `NetworkHelper.SetupAndConnectNetwork()` more than once, or add an option to retry.

josesimoes, 2023-09-29:
This is by design and on purpose.
Going through the network setup sequence once it's connected causes unpredictable behavior. 
Plus,  there are steps in the network setup that need to occur sequentially. 
Therefore this check to prevent it from happening.

It's easier to reboot the damm thing than to try reconnecting and/or restoring the connection. 😉

networkfusion, 2023-09-29:
We are pretty screwed on an STM32 in that instance as there is no way of rebooting though the current managed interfaces... (e.g. `nanoFramework.Hardware.Stm32.Power.RebootDevice();` does not exist).

josesimoes, 2023-09-29:
I beg to differ.!
Fortunately someone thought about such requirement and there is:
https://docs.nanoframework.net/api/nanoFramework.Runtime.Native.Power.html#nanoFramework_Runtime_Native_Power_RebootDevice 😉

networkfusion, 2023-09-29:
Fair enough (I was looking for that, but nothing in samples et. al.) I will close this for the moment.

closed 2023-09-29 by networkfusion