nanoframework/Home #1377: Resource manager / generator should support multiple resource files in a single project

CoryCharlton, 2023-10-04:
### Description

The resource manager / resource generator should properly handle multiple resource files in a single project.

** OR **

The resource generator should provide a clear error explaining that only a single resource file per project is supported if that is a limitation that should be enforced.

### How to solve the problem

Resolve the issues related to multiple resource files OR provide a clear error message and build failure stating that only a single resource file is supported.

### Aditional context

I created a quick POC that reproduces these issues at https://github.com/CoryCharlton/nanoFramework.ResourceManager.Issues

Ellerbach, 2023-12-26:
I've been starting to look at all this and for the first case where you have same name in 2 different resources, it's not going to be very easy to find before compilation. The error could be improved maybe.

For the other cases, I have additional questions:

* How do you force having positive numbers? I only get negative ones
* I was not able to reproduce by creating the project from scratch the behavior you describe in th 2 other cases (so not the same resource name). Do you make any specific change to get to those edge cases?

CoryCharlton, 2023-12-28:
> * How do you force having positive numbers? I only get negative ones

The resource ids appeared to be directly related to the resource names so I just played around by adding "1", "2", etc. until I got a value that would reproduce the issue.

> * I was not able to reproduce by creating the project from scratch the behavior you describe in th 2 other cases (so not the same resource name). Do you make any specific change to get to those edge cases?

I don't recall having to do anything special other than adding multiple resource files.

josesimoes, 2024-01-02:
Resources start to "exist" at build time. The current MDP version was designed for a single resource per project. To be honest I don't think it's worth the effort to put work on this and make a bunch on changes to make it happen. It's by design, so let's accept this.

Ellerbach, 2024-03-26:
@CoryCharlton I'm moving this as a discussion as I don't think the work to be done is worse the effort in most scenarios. Also, it's something that can be properly documented.
And last, we're now adding file storage deployment from Visual Studio and nanoff. So in a project, you can mark some content file as to deploy and they'll be deployed in a storage. Currently only internal storage is supported on few devices. That should also help when it's about deploying files in a storage rather than in the resources