nanoframework/Home #1380: `WirelessAPConfiguration` should validate `MaxConnections`

CoryCharlton, 2023-10-09:
### Library/API/IoT binding

System.Net

### Description

`WirelessAPConfiguration` has a `ValidateConfiguration()` method but it does not check `MaxConnections` so it's possible to set this to zero (maybe even a negative number).

This allows the process to appear to be working fine but fails to allow clients to connect and retrieve and IP address.

### How to reproduce

```
wirelessAPConfiguration.MaxConnections = 0;
wirelessAPConfiguration.SaveConfiguration();
```

### Expected behaviour

An error is throw on invalid configuration.

CoryCharlton, 2023-10-09:
Actually all the existing tests are currently skipped:

```
[Setup]
public void SetupConnectToEthernetTests()
{
    // Comment next line to run the tests on a real hardware
    Assert.SkipTest("Skipping tests using nanoCLR Win32 in a pipeline");
}
```

As such I'll submit the PR without tests and await feedback.

josesimoes, 2023-10-09:
> Actually all the existing tests are currently skipped:

This is happening because Unit Tests, by default, run on virtual device, which doesn't implement network.
For the Unit Test to run locally on a HARDWARE device you have to edit the .runsettings file and set `<IsRealHardware>True</IsRealHardware>`. Assuming you have ONE nano device connected the Unit Tests will run there. 😉

closed 2023-10-10 by josesimoes