nanoframework/Home #1382: Support deserializing/serializing camelCase JSON fields to PascalCase properties

CoryCharlton, 2023-10-10:
### Description

Support deserializing/serializing camelCase JSON fields to PascalCase properties. 

camelCase seems to be the preferred casing for JSON fields.

### How to solve the problem

The simplest (and potentially most memory / processing efficient) option would be to support a JsonPropertyName attribute that allows complete control over the Property <-> JSON field mapping. That being said it sounds like there might be a limitation in the existing nanoFramework reflection that may prevent this solution.

Another option (for deserialization) would be to check for a direct match like the code does now and then fallback to checking for a match where the first letter of the property name is lower cased.

For serialization it would be as straight forward as defaulting to lowering the first letter.

### Describe alternatives you've considered

As a workaround/hack I would create copies of my DTOs with camelCased property names and map between them :yuck:

```
class RealObject 
{
    public string FieldWithValue { get; }
}
```

```
class FakeObjectForSerialization
{
    public string fieldWithValue { get; }
}
```

josesimoes, 2023-10-10:
I believe we already have support for this. 
https://docs.nanoframework.net/api/nanoFramework.Json.Configuration.Settings.html#nanoFramework_Json_Configuration_Settings_CaseSensitive

CoryCharlton, 2023-10-10:
My bad. I replied on typing 'Js' in the IDE and saw no configuration related options.

Next time I'll check the docs 👍

josesimoes, 2023-10-10:
No worries! It happens.

closed 2023-10-10 by josesimoes