nanoframework/Home #1383: Transient services are instantiated twice while creating a service that depends on them

CoryCharlton, 2023-10-10:
### Library/API/IoT binding

nanoFramework.DependencyInjection

### Description

Transient services are instantiated twice while creating a service that depends on them

### How to reproduce

Execute the same code below

### Expected behaviour

Dependencies are only created a single time while resolving the service that depends on them.

### Sample project or code

public class Program
{
    public static void Main()
    {
        var serviceCollection = new ServiceCollection();
        serviceCollection.AddSingleton(typeof(ISingletonService), typeof(SingletonService));
        serviceCollection.AddTransient(typeof(ITransientService), typeof(TransientService));

        var serviceProvider = serviceCollection.BuildServiceProvider();

        var singletonService = (ISingletonService) serviceProvider.GetService(typeof(ISingletonService));
        var transientService = (ITransientService) serviceProvider.GetService(typeof(ITransientService));

        Thread.Sleep(Timeout.Infinite);
    }
}

public interface ISingletonService
{
    Guid Id { get; }
}

public interface ITransientService
{
    Guid Id { get; }
}

public class SingletonService : ISingletonService
{
    public Guid Id { get; } = Guid.NewGuid();

    public SingletonService(ITransientService transientService)
    {
        Debug.WriteLine($"{nameof(SingletonService)}.ctor - {Id} - TransientService: {transientService.Id}");
    }
}

public class TransientService : ITransientService
{
    public Guid Id { get; } = Guid.NewGuid();

    public TransientService()
    {
        Debug.WriteLine($"{nameof(TransientService)}.ctor - {Id}");
    }
}

### Aditional information

The reason I believe this is an issue is that if there is any instantiation logic (ie: starting a `Timer`) in the constructor of the transient dependency it will be executed and the resolved service will be thrown away.

CoryCharlton, 2023-10-10:
The issue appears to be occurring while resolving the parameters to use for the constructor:

https://github.com/nanoframework/nanoFramework.DependencyInjection/blob/f43b8b3b77b63248768553d7ead7ca77ed56fdd2/nanoFramework.DependencyInjection/DependencyInjection/ServiceProviderEngine.cs#L266

I'll keep digging but I wonder if it's possible to check if the parameter can be resolved without actually creating an instance of it.

Additionally it looks like an optimization would be to break out of the loop when we set `length = -1` since we know that constructor can't be used at that point.

CoryCharlton, 2023-10-10:
It looks like we could check that a `ServiceDescriptor` has been registered to accomplish the same results.

```
private ServiceDescriptor[] GetServiceDescriptors(Type serviceType, IServiceCollection scopeServices = null)
{
    var serviceDescriptors = new ArrayList();

    if (scopeServices is not null)
    {
        foreach (ServiceDescriptor descriptor in scopeServices)
        {
            if (descriptor.ServiceType != serviceType) continue;

            serviceDescriptors.Add(descriptor);
        }
    }

    foreach (ServiceDescriptor descriptor in Services)
    {
        if (descriptor.ServiceType != serviceType) continue;
        
        serviceDescriptors.Add(descriptor);
    }

    return (ServiceDescriptor[])serviceDescriptors.ToArray(typeof(ServiceDescriptor));
}
```

And in `GetParameters:

```
foreach (ParameterInfo parameter in parameters)
{
    Type type = parameter.ParameterType;

    if (type.IsResolvable())
    {
        // check for simple binding first
    }
    else
    {
        var serviceDescriptors = GetServiceDescriptors(type);
        if (serviceDescriptors is null || serviceDescriptors.Length <= 0)
        {
            // binding could not be resolved ignore constructor
            length = -1;
            break;
        }

    }
}
```

I'm going to send over an in progress PR shortly to start getting code review.

josesimoes, 2023-10-10:
@CoryCharlton let's have that PR! We 💜  PRs! 😉

CoryCharlton, 2023-10-10:
@josesimoes I said it would be an in progress PR but the solution ended up being more straightforward than I expected 👍

closed 2023-11-06 by josesimoes