nanoframework/Home #1384: ESP32 Crash when connecting to Wifi

alberk8, 2023-10-14:
### Target name(s)

ESP32_PSRAM_REV3

### Firmware version

1.8.1.633

### Was working before? On which version?

1.8.1.629

### Description

The system Panic upon connecting to wifi.

### How to reproduce

```
 CancellationTokenSource cancelToken = new CancellationTokenSource(40_000);
 bool isConnected = WifiNetworkHelper.ConnectDhcp(_ssid, _password, WifiReconnectionKind.Automatic, requiresDateTime: false, token: cancelToken.Token);
```

### Expected behaviour

No crash

### Aditional information

Backtrace

```
0x401b6664: HAL_AssertEx at /workspaces/nf-interpreter/targets/ESP32/_common/targetHAL.c:11
0x400d50a5: CLR_RT_HeapBlock::Assign(CLR_RT_HeapBlock const&) at /workspaces/nf-interpreter/src/CLR/Include/nanoCLR_Runtime__HeapBlock.h:1229
0x400e2649: CLR_RT_Thread::Execute_IL(CLR_RT_StackFrame&) at /workspaces/nf-interpreter/src/CLR/Core/Interpreter.cpp:1107
0x400e22a6: CLR_RT_Thread::Execute_Inner() at /workspaces/nf-interpreter/src/CLR/Core/Interpreter.cpp:843
0x400e1ffd: CLR_RT_Thread::Execute() at /workspaces/nf-interpreter/src/CLR/Core/Interpreter.cpp:645
0x400dce11: CLR_RT_ExecutionEngine::ScheduleThreads(int) at /workspaces/nf-interpreter/src/CLR/Core/Execution.cpp:1238
0x400dc599: CLR_RT_ExecutionEngine::Execute(wchar_t*, int) at /workspaces/nf-interpreter/src/CLR/Core/Execution.cpp:665
0x400f592a: ClrStartup at /workspaces/nf-interpreter/src/CLR/Startup/CLRStartup.cpp:407
0x4011aff5: CLRStartupThread at /workspaces/nf-interpreter/targets/ESP32/_nanoCLR/CLR_Startup_Thread.c:17
0x40119235: main_task at /workspaces/nf-interpreter/targets/ESP32/_IDF/esp32/app_main.c:37
```

CoryCharlton, 2023-10-14:
I'm also experiencing this on ESP32_PSRAM_BLE_GenericGraphic_REV3 1.8.1.633.

alberk8, 2023-10-14:
Bluetooth is also not working (crash) on ESP32_BLE_REV3.

networkfusion, 2023-10-14:
Just for reference, the changes between the versions were:

[image]

josesimoes, 2023-10-15:
Yes, everything that is fixed and broken is on me. 😁

closed 2023-10-18 by josesimoes