nanoframework/Home #1390: Bug in Hashtable (or CLR_RT_HeapBlock::ObjectsEqual)

MateuszKlatecki, 2023-10-27:
### Target name(s)

ALL

### Firmware version

1.8.1.685

### Description

Attempting to run the code below results in the device hanging.

```cs
using System.Collections;
using System.Diagnostics;
using System.Threading;

namespace HashtableProblem
{
    public class SomeElement
    {

    }
    public class Program
    {
        public static void Main()
        {
            Thread.Sleep(10000);

            Debug.WriteLine("Hello from nanoFramework!");

            Hashtable ht = new Hashtable();

            for (ulong i = 1; i <= 100; i++)
            {
                SomeElement se = new SomeElement();
                Debug.WriteLine($"Adding element no {i}");
                ht.Add(se, i);
            }

            Thread.Sleep(Timeout.Infinite);

        }
    }
}
```

The problem occurs when the [ObjectsEquals](https://github.com/nanoframework/nf-interpreter/blob/main/src/CLR/Core/CLR_RT_HeapBlock.cpp#L1227) method is called for the first time in the [Hashtable::InsertNative](https://github.com/nanoframework/nf-interpreter/blob/main/src/nanoFramework.System.Collections/nf_system_collections_System_Collections_Hashtable.cpp#L261) method.

The `leftDataType` and `rightDataType`  variables have the values DATATYPE_CLASS and it looks like in CLR_RT_HeapBlock::ObjectsEqual method is missing such a case and the firmware stops at this [assert](https://github.com/nanoframework/nf-interpreter/blob/main/src/CLR/Core/CLR_RT_HeapBlock.cpp#L1385
)
 

### How to reproduce

Run code from description

TekuSP, 2023-10-29:
I can confirm this bug, for me it happens when running WPF Simple Sample, with GC9A01 display. It surely must worked in the past, or I dont know how this example got written.

networkfusion, 2023-10-30:
Just for (my) clarity at this point... Something has changed at some point that has (at very least) broken the WPF sample. I think that it should be a higher priority for that,

josesimoes, 2023-10-30:
The WPF sample got broken because there are now object being used as keys in a Hastable. See [here](https://github.com/nanoframework/nanoFramework.Graphics/blob/0bc61c88556edd8fe943b285cb7357f7c52b13ec/nanoFramework.Graphics/Presentation/UIElement.cs#L171) and there are other places.

I would love to make this a higher priority but it's beyond me the workload we're currently dealing with... 🤷🏻‍♂️

TekuSP, 2023-10-30:
I think graphics are huge selling point why to use Nanoframework, and I think it should have higher priority.

networkfusion, 2023-10-30:
> I would love to make this a higher priority but it's beyond me the workload we're currently dealing with... 🤷🏻‍♂️

lol, I am hoping this is a nF priority, rather than yours and someone else may pick it up 😅

TekuSP, 2023-10-31:
> The WPF sample got broken because there are now object being used as keys in a Hastable. See [...]

By blame that code (WPF) is 4 years old. What changed in Hasthtable in 4 years so it does not work now? Or WPF Sample never worked?

josesimoes, 2023-10-31:
Not sure if I'm following your point... Wpf sample was working until those changes came. If you look at the Wpf code you can check the history. By all means don't trust me and check by yourself.

TekuSP, 2023-10-31:
@josesimoes By blame in UIElement
[image]

Hashtable is used that way for 4 years. What changed now?

CoryCharlton, 2023-11-16:
Removing a key that doesn't exist (my use case was `byte` as the key type) also seems to crash the device (haven't been able to debug beyond that).

josesimoes, 2023-12-01:
Interestingly, adding a Unit Test with the cod above, seems to work... 😯 🤔 
nanoframework/System.Collections#123

closed 2023-12-01 by josesimoes