nanoframework/Home #1392: Esp32 Cant Connect to WIFI

AlirezaP, 2023-10-29:
### Target name(s)

ESP32_PSRAM_REV0

### Firmware version

1.8.1.620

### Description

always connection fail.
If I understand correctly the problem, rise from "GetNativeScanReport" in "WifiAdapter.cs".

it worked with ESP32_WROOM_32 and windows.device.wifi but now i update the Firmware to ESP32_PSRAM_REV0 with System.Device.WiFi NuGet and it not work.!

### How to reproduce

bool result = WifiNetworkHelper.ScanAndConnectDhcp(Ssid, Password, WifiReconnectionKind.Automatic, false, 0, cs.Token);

### Expected behaviour

connect successfully

### Screenshots

**img1**
[image]

**img2**
[image]

alberk8, 2023-10-30:
The issue is using the PSRAM firmware on a **non PSRAM** hardware.   The ESP32 WROOM does not have PSRAM.   Please use the ESP32_REV0 firmware.

AlirezaP, 2023-10-30:
Thank you. Yes, you are right. esp32 wroom 32 does not have PSRAM. I changed firmware to ESP32_REV0. The exception was changed, it changed to "unspecific Failor". it rise from "NativeConnect" method in "WifiAdapter.cs".

AlirezaP, 2023-10-30:
[image]

alberk8, 2023-10-31:
> [image]

How about doing a [Wifi Scan](https://github.com/nanoframework/Samples/tree/main/samples/Wifi/ScanWiFi)?  Do you get a return that shows the intended SSID that you want to connect?

AlirezaP, 2023-10-31:
Wifi scan works. It lists all available access points. But when connecting, the exception is raised

AlirezaP, 2023-10-31:
[image]

FreeFrags, 2023-12-02:
I can confirm that i have this issue on a: ESP32 (ESP32-D0WD-V3 (revision v3.1)) too. 

I flashed esp32_ble_rev3 and tried  esp32_ble_rev0 too (AlbertK advised me to try this in Discord). 

Im trying to run the improvWifi sample, the following line returns false:
https://github.com/nanoframework/Samples/blob/4b1754686fde18562062c1c7f9a846769c34d1db/samples/Bluetooth/ImprovWifi/Improv.cs#L529

Hopefully im not adding this to a different issue but the symptoms and the exceptions look very similar.

Ellerbach, 2024-03-26:
@FreeFrags the improv sample has been improved. So, should work better.

@AlirezaP do you still have the issue? I've been recently using it on a ESP32 REV0 without PSRAM and all works fine.

networkfusion, 2024-05-09:
Please let us know if this is still an issue.

torbacz, 2024-05-09:
Testes with ESP32_REV0-1.9.1.93, works.

closed 2024-05-09 by networkfusion