nanoframework/Home #1393: Web request fails when reading response

oleneveu, 2023-10-29:
### Library/API/IoT binding

System.Net / System.Net.Http

### Visual Studio version

17.7.6

### .NET nanoFramework extension version

2022.3.0.65

### Target name(s)

ESP32_REV3

### Firmware version

1.8.1.685

### Description

* I send a request to a local http server through http (not https).
* The server receives the request and sends the response.
* When the response processed by the nanoframework, an exception is thrown.

The behavior is the same when using HttpClient and HttpWebRequest.
It is also the same for every endpoint of the server I have tried.

I have tried to send the request through an http proxy (Telerik Fiddler) to that this is not related to the server. The proxy receives and forwards both the request and the response with no issue.

I have tried older firmware versions down to 1.8.1.1 with no luck.

Below is a minimalistic sample code to reproduce the issue.

### Sample project or code

```C#
public static void Main()
{
    CancellationTokenSource cs = new(60000);
    bool success = WifiNetworkHelper.ConnectDhcp(Configuration.WifiSSID, Configuration.WifiPassword, requiresDateTime: true, token: cs.Token);
    Debug.WriteLine($"Network address: {NetworkInterface.GetAllNetworkInterfaces()?[0]?.IPv4Address}");

    var httpClient = new HttpClient()
    {
        BaseAddress = new Uri($"http://{Configuration.HueBridgeAddress}/api/{Configuration.HueUserToken}/"),
        Timeout = TimeSpan.FromSeconds(5),
    };

    try
    {
        string response = httpClient.GetString($"lights/1");
        Debug.WriteLine($"Response: {response}");
    }
    catch (Exception ex)
    {
        Debug.WriteLine(ex.Message);
    }

    Thread.Sleep(Timeout.Infinite);
}
```

**Output:**
```
Network address: 192.168.2.230
    ++++ Exception System.Net.Sockets.SocketException - CLR_E_FAIL (1) ++++
    ++++ Message: 
    ++++ System.Net.Sockets.NativeSocket::recv [IP: 0000] ++++
    ++++ System.Net.Sockets.Socket::Receive [IP: 0018] ++++
    ++++ System.Net.Sockets.NetworkStream::Read [IP: 0062] ++++
    ++++ System.Net.InputNetworkStreamWrapper::ReadInternal [IP: 00d7] ++++
    ++++ System.Net.InputNetworkStreamWrapper::Read [IP: 000d] ++++
    ++++ System.Net.Http.StreamContent::SerializeToStream [IP: 0061] ++++
    ++++ System.Net.Http.HttpContent::LoadIntoBuffer [IP: 002a] ++++
    ++++ System.Net.Http.HttpClient::Send [IP: 006f] ++++
    ++++ System.Net.Http.HttpClient::Get [IP: 000c] ++++
    ++++ System.Net.Http.HttpClient::GetString [IP: 0006] ++++
    ++++ HueCommand.Program::Main [IP: 005c] ++++
Exception thrown: 'System.Net.Sockets.SocketException' in System.Net.dll
Exception was thrown: System.Net.Sockets.SocketException
```

alberk8, 2023-10-30:
Can you check for the WIFI connectivity is good before attempting to connect to HttpClient ?

```
if (success)
{
     // do you http get here.
}
```

oleneveu, 2023-10-30:
I have provided a sample that reproduces the issue, but the actual code checks the WIFI connectivity.
Either way, the connectivity is good as I can see the both the request and response when monitoring the network.

The sample code performs only one request, but I have tested sending requests repeatedly and the result is always the same.
This is not a random error, I have not been able to send a single request successfully (or more precisely, to receive a single response as the request is always sent properly).

I have tried refactoring the code in any possible way I can imagine without luck.

oleneveu, 2023-10-30:
I'm closing that issue as after further investigation it appears that it only happens in specific cases that I am not able to identify yet.
I will post another issue when I have figured out a way to reproduce it in a consistent way.

closed 2023-10-30 by oleneveu