nanoframework/Home #1395: [System.Math] `Math.Max(float)` and `Math.Min(float)` are 3x slower than the `double` overloads

CoryCharlton, 2023-11-03:
### Library/API/IoT binding

System.Math

### Description

While benchmarking the System.Math function I noticed that  [`Math.Max(float)`](https://github.com/nanoframework/System.Math/blob/f516ba57ebef25c4d13e49717d4a831f6567549e/nanoFramework.System.Math/Math.cs#L376) and [`Math.Min(float)`](https://github.com/nanoframework/System.Math/blob/f516ba57ebef25c4d13e49717d4a831f6567549e/nanoFramework.System.Math/Math.cs#L409) are 3x slower than [`Math.Max(double)`](https://github.com/nanoframework/System.Math/blob/f516ba57ebef25c4d13e49717d4a831f6567549e/nanoFramework.System.Math/Math.cs#L367) and [`Math.Min(double)`](https://github.com/nanoframework/System.Math/blob/f516ba57ebef25c4d13e49717d4a831f6567549e/nanoFramework.System.Math/Math.cs#L400C19-L400C19).

While reviewing the code I see that both of the float overloads just delegate down to the double overload and cast the result to a float. I'm assuming this is the root cause of the performance difference.

### How to reproduce

Run the benchmarks at https://github.com/CoryCharlton/CCSWE.nanoFramework.Math

### Expected behaviour

Similar, if not faster, performance.

### Aditional information

```
Console export: MaxBenchmarks benchmark class.
| MethodName        | ItterationCount | Mean                  | Min   | Max   |
| --------------------------------------------------------------------------- |
| Max_Double_System | 250             | 13.12 ms              | 10 ms | 20 ms |
| Max_Float_System  | 250             | 29.32 ms              | 20 ms | 40 ms |
```

```
Console export: MinBenchmarks benchmark class.
| MethodName        | ItterationCount | Mean                  | Min   | Max   |
| --------------------------------------------------------------------------- |
| Min_Double_System | 250             | 11.2 ms               | 0 ms  | 20 ms |
| Min_Float_System  | 250             | 35.64 ms              | 20 ms | 50 ms |
```

CoryCharlton, 2023-11-03:
I have no problem taking a stab at fixing/implementing this (well aside from getting a build environment setup) but I wanted to make sure I wasn't missing something regarding why the float overloads were not implemented on the native side.

Ellerbach, 2023-11-04:
Yes, native can be slower than pure manage in some cases. And yes, if you can make the PR, even better! Thanks a lot for the investigation. I guess it would be good to implement all in manages for both float and double and see when native is faster and when it's slower! And then make the best out of all this!

closed 2023-11-09 by josesimoes