nanoframework/Home #1397: float.IsNegativeInfinity() and double.IsNegativeInfinity() do not correctly identify negative infinity

CoryCharlton, 2023-11-05:
### Library/API/IoT binding

PAL?

### Description

I noticed several unit tests in System.Math we failing (before I touched it, I swear ;-)). The root cause appears to be that IsNegativeInfinity returns false even though the parameter is negative infinity.

The code in [System::Double::IsNegativeInfinity](https://github.com/nanoframework/nf-interpreter/blob/c15fcd62a6e8858ce02f85f95f861f4403552830/src/PAL/Double/nanoPAL_NativeDouble.cpp#L135) looks correct and I reproduced the logic in an online compiler https://godbolt.org/z/5PWKhhf81 (although not everything is the same)

Given that the code is correct, and the debugger shows the correct value before executing the method, I have to assume something is changing the parameter in between the caller and the callee 🤷 

### How to reproduce

Run the sample code (see screenshot below for results)

### Expected behaviour

IsNegativeInfinity() returns true when value is negative infinity.

### Screenshots

[image]

### Sample project or code

```
double d_nan = 0.0 / 0.0;
double d_pos_inf = (3.0 / 0.0);
double d_neg_inf = (-3.0 / 0.0);

Console.WriteLine($"Expect d_nan={d_nan} - {double.IsNaN(d_nan)}, d_pos_inf={d_pos_inf} - {double.IsPositiveInfinity(d_pos_inf)}, d_neg_inf={d_neg_inf} - {double.IsNegativeInfinity(d_neg_inf)}");

float f_nan = 0.0f / 0.0f;
float f_pos_inf = (3.0f / 0.0f);
float f_neg_inf = (-3.0f / 0.0f);

Console.WriteLine($"Expect f_nan={f_nan} - {float.IsNaN(f_nan)}, f_pos_inf={f_pos_inf} - {float.IsPositiveInfinity(f_pos_inf)}, f_neg_inf={f_neg_inf} - {float.IsNegativeInfinity(f_neg_inf)}");
```

josesimoes, 2023-11-06:
> I noticed several unit tests in System.Math we failing (before I touched it, I swear ;-)). 

That's curious... because these run on the AZDO pipeline last May (last time the lib was updated) and they all [run fine}(https://dev.azure.com/nanoframework/System.Math/_build/results?buildId=42474&view=logs&j=12f1170f-54f2-53f3-20dd-22fc7dff55f9&t=7bfb7e0f-549d-5929-c603-6e8046613b9e)... 😯

CoryCharlton, 2023-11-06:
> That's curious... because these run on the AZDO pipeline last May (last time the lib was updated) and they all [run [...]

Curious indeed. That gives me a point in history to start investigating from (although a very wide point, maybe an integration test suite would be useful here 🤷 )

josesimoes, 2023-11-06:
If you revert to the commit just before your changes, any of the Unit Tests fail?

CoryCharlton, 2023-11-06:
Yes these same tests were failing before I started making any changes. I just switched back to main and reconfirmed that they fail with the latest firmware version/emulator.

[image]

closed 2025-02-06 by josesimoes